ros2-dotnet/ros2_dotnet #1: Patch files?

calvertdw, 2018-07-23:
The patch file in the README is missing: `rosidl_typesupport_ros2_dotnet.patch`

Additionally, the patch file in the root (`rosidl_typesupport_ros2_dotnet.patch`) patches a file that does not seem to exist: `rosidl_default_generators/CMakeLists.txt``

calvertdw, 2018-07-23:
I get the error:
```
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "Findrosidl_default_generators.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "rosidl_default_generators", but CMake did not find one.

  Could not find a package configuration file provided by
  "rosidl_default_generators" with any of the following names:

    rosidl_default_generatorsConfig.cmake
    rosidl_default_generators-config.cmake

  Add the installation prefix of "rosidl_default_generators" to
  CMAKE_PREFIX_PATH or set "rosidl_default_generators_DIR" to a directory
  containing one of the above files.  If "rosidl_default_generators" provides
  a separate development package or SDK, be sure it has been installed.
```

esteve, 2018-07-24:
I've pushed https://github.com/esteve/ros2_dotnet/tree/remove-patches. I'm reenabling AppVeyor, but in the meantime, do you mind testing it locally and see if it works? Thanks.

calvertdw, 2018-07-24:
I get the same error. I'll try purging and building again.

calvertdw, 2018-07-24:
Sorry I just realized you di not merge that branch. Let me chack that out.

calvertdw, 2018-07-24:
Needs package format version 3: https://github.com/esteve/ros2_dotnet/blob/e0ea001d2a63a6d23d440aed26f9223dc5459d2a/rosidl_generator_dotnet/package.xml#L2

esteve, 2018-07-24:
Fixed, crossing my fingers :-) https://ci.appveyor.com/project/esteve/ros2-dotnet/build/1.0.6

calvertdw, 2018-07-24:
I now get the same error as here: https://github.com/esteve/ros2_java/issues/49

calvertdw, 2018-07-24:
Odd yours failed at fastrtps but mine got past that.

esteve, 2018-07-27:
I finally got past Fast-RTPS on AppVeyor, but it failed with `rcl_yaml_param_parser` (see https://ci.appveyor.com/project/esteve/ros2-dotnet/build/1.0.25). It seems that MSVC on AppVeyor is a bit strict and causes some warnings to be labeled as error, I'll submit a PR to fix them.

esteve, 2018-08-01:
I've just merged `remove-patches` and set up AppVeyor to run every night.

esteve, 2018-08-04:
Closing this since CI passes now, let me know if it works for you and feel free to reopen the ticket if it doesn't. Thanks!

closed 2018-08-04 by esteve