ros2-dotnet/ros2_dotnet #2: ros2_dotnet on Unity: UnsatisfiedLinkError

dvargasfr, 2018-10-16:
Hi! I'm experimenting an error trying to execute a simple talker/listener example on Unity. The talker/listener code is adapted to Unity Start()/Update() execution model. I succesfully compiled ros2_dotnet for Windows and included the appropriate .dll's (those from install\bin folder and others needed from install\Lib) to a Plugins folder. And at runtime get this UnsatisfiedLinkError: 

UnsatisfiedLinkError: rcldotnet_native.dll
ROS2.Utils.DllLoadUtilsUWP.ROS2.Utils.DllLoadUtils.LoadLibrary (System.String fileName) (at <ab710c4a68b44e538b236682ed3a97a2>:0)
ROS2.RCLdotnetDelegates..cctor () (at <f5d1f0387ca84a3aa9ff3108adbc6d1a>:0)
Rethrow as TypeInitializationException: The type initializer for 'ROS2.RCLdotnetDelegates' threw an exception.
ROS2.RCLdotnet.Init () (at <f5d1f0387ca84a3aa9ff3108adbc6d1a>:0)
rcldotnettalker.Start () (at Assets/Scripts/rcldotnettalker.cs:23)

Maybe have dll's been generated incorrectly for using them on Unity? Maybe are they correctly generated but not correctly referenced just placing them in Plugins folder?

dvargasfr, 2018-10-22:
@esteve I found that there are a lot of missing DLL dependencies for rcldotnet_native.dll. I got it using the tool Dependency Walker.
[image]

esteve, 2018-10-23:
I don't know for certain, but it'd be strange to put non-plugin DLLs in the Plugins directory. Do you know where in the documentation for Unity states where to put dependency DLLs? I can't seem to find it.

> I found that there are a lot of missing DLL dependencies for rcldotnet_native.dll. I got it using the tool Dependency Walker.

Yes, you'll have to copy those from Windows\System manually. The CMake scripts do that for you, but since you're using Unity directly, it can't be done automatically.

Also, have you tried building a simple UWP instead? It's just to make sure that it all works for you without adding the complexity of Unity.

dvargasfr, 2018-10-23:
> Yes, you'll have to copy those from Windows\System manually. The CMake scripts do that for you, but since you're using Unity directly, it can't be done automatically.

Dependecy Walker scans dependencies in my own Windows system. I mean, it is no related to Unity, all of those missing DLL aren´t in my system. Should they be there? Any install forgotten?

> Also, have you tried building a simple UWP instead? It's just to make sure that it all works for you without adding the complexity of Unity

I'm currently working on deploying Publisher/Subscriber in HoloLens, but as you suggest I'll try to build a simpler UWP example without using Unity. 
BTW one thing here: I am developing on a 64bits Windows10, and Unity provides the option 'Target Device: HoloLens' when building a UWP app. Despite this Unity configuration, should I use uwp-32 or uwp-64 compilation to deploy ROS2 on HoloLens?

dvargasfr, 2018-11-04:
After hard research works using serveral Windows diagnostic tools (the most usefull Process Monitor https://docs.microsoft.com/en-us/sysinternals/downloads/procmon and Visual Studio gflags for more verbose debugging information https://stackoverflow.com/questions/36315538/how-to-determine-which-dll-dependency-is-failing-to-load-in-windows-store-univer ) I was able to overcome some issues: 
- Problems with tinyxml2.dll. Solved installing newest version.
- mrt100_app.dll and SharedLibrary.dll couldn't be found because C:\Program Files\WindowsApps was inaccessible due to access permission.
- Right include of DLLs in the VS UWP app, https://social.msdn.microsoft.com/Forums/en-US/26013e5e-3ff1-44a0-8797-0474dabd12d0/uwpincluding-dlls-in-a-uwp-application-in-visual-studio?forum=wpdevelop .

Simple ROS2 talker/listener UWP app is not running yet, but I think it is close. I'll let you know when I achieve it.

MischaRo, 2019-01-16:
Hello @dvargasfr , I am also interested in running ROS2 on Unity.

 After compiling ros2_dotnet for Windows  I included the dlls and libs from the install install\bin and install\Lib folder to Assets/Plugins folder in Unity
I created a script to work as a ROS2 listener, but unfortunately Unity does not seem to recognize the included dlls, as `using ROS2; ` or any other ROS2 method is not recognized.

Did you run into any similar problems when you created your project?

 
best regards

samiamlabs, 2019-01-19:
Hi.

Any progress @dvargasfr?

I made a C# wrapper for the crystal release of ROS2 targeted at use with Unity. Got it working on Linux but haven't tested it on Windows yet.
(https://github.com/DynoRobotics/unity_ros2)

Made a temporary fix in dotnet_cmake_module because I got the wrong _target_path for Mono when installing the DLLs on Linux for some reason.
If you use the master branch of dotnet_cmake_module that should not be a problem.

Linking to c-libraries on Linux seems to work as long as ```LD_LIBRARY_PATH``` is set to the proper paths in the terminal you use to start Unity (by sourcing setup.bash beforehand).

I may get around to testing it on Windows myself at some point as well.

dvargasfr, 2019-01-21:
Hi @samiamlabs ,

> I may get around to testing it on Windows myself at some point as well.

FYI, some details about ROS2 UWP apps development are being documentend here https://github.com/IntelligentRoboticsLabs/ARViz

MischaRo, 2019-01-23:
Hi, @dvargasfr
I am also interested in using ros2_dotnet for Unity and I was closely following this thread. Unfortunately I was not able to get it to work so far.

What I tried till now:

I downloaded the release build `ros2-dotnet-uwp-x86` and copied all the dlls from `ros2> install> bin` to the Unity folder `Assets> Plugins> x86`
After that I used dependencies tool (https://github.com/lucasg/Dependencies) https://github.com/lucasg/Dependencies to find all the missing DLLs and put them in the same folder.

After thatI made a small script:

```
using UnityEngine;
using ROS2;
public class Ros2test : MonoBehaviour

{
    string[] args;
    void Start()
    {
        RCLDotnetTalker.Main(args);
    }
}
```

but `RCLDotnetTalker` won’t even be recognized.

Then I used this script:

```
using UnityEngine;
using System;
using System.Runtime.InteropServices;

using ROS2;

public class Ros2test : MonoBehaviour {

string[] args;
    // Use this for initialization
    void Start () {
 RCLdotnet.Init();
        INode node = RCLdotnet.CreateNode("listener");
        RCLdotnet.Spin(node);
    }
	
	
}
}
```

And I could see that RCLdotnet, INode, and RCLdotnet were actually recognized.
But when I would start unity I would get an

```
UnsatisfiedLinkError: rcldotnet_native.dll
ROS2.Utils.DllLoadUtilsUWP.ROS2.Utils.DllLoadUtils.LoadLibrary (System.String fileName) (at <a971252f0a0c46c886d8c22d055045cc>:0)
ROS2.RCLdotnetDelegates..cctor () (at <a26c7cefc8154ec59e5cd7c6b74c18dc>:0)
Rethrow as TypeInitializationException: The type initializer for 'ROS2.RCLdotnetDelegates' threw an exception.
ROS2.RCLdotnet.Init () (at <a26c7cefc8154ec59e5cd7c6b74c18dc>:0)
Ros2test.Start () (at Assets/Scripts/Ros2test.cs:24)

```

I tried to reference the in the script directly with:
```
[DllImport("rcldotnet_native")]

    private static extern int native_rcl_init();
.
.
.
int test = native_rcl_init();
```

But the I get a DllNotFoundException: rcldotnet_native exception.

I don´t really know what else to do.

Could you maybe share the method that you used to get it working?
That would be great :)

dvargasfr, 2019-01-24:
Hi @MischaRo ,

Please keep checking https://github.com/IntelligentRoboticsLabs/ARViz , every detail you are asking for is going to be explained over there. Throughout today "Unity" aspects are going to be updated.
If you find any doubt after that, please write your issue on that repo.

MischaRo, 2019-01-24:
Hi @dvargasfr ,

Thanks for the reply, I will be closely watching the updates and give you feedback if I have any problems

dvargasfr, 2019-01-24:
@esteve sure! I include it asap.