ros2-dotnet/ros2_dotnet #5: Unable to run rcldotnet_examples

dvargasfr, 2018-11-13:
Hi @esteve , I installed everything from sources, only test_msgs skiped, but I get this error when I try to run the talker/listener example on a Windows10 cmd:

```
C:\dev\ros2_dotnet_ws>ros2 run rcldotnet_examples rcldotnet_talker

Unhandled Exception: System.TypeInitializationException: The type initializer for 'ROS2.RCLdotnetDelegates' threw an exception. ---> ROS2.Utils.UnsatisfiedLinkError: rcldotnet_native.dll
   at ROS2.Utils.DllLoadUtilsUWP.ROS2.Utils.DllLoadUtils.LoadLibrary(String fileName) in C:\dev\ros2_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet_common\DllLoadUtils.cs:line 152
   at ROS2.RCLdotnetDelegates..cctor() in C:\dev\ros2_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet\RCLdotnet.cs:line 104
   --- End of inner exception stack trace ---
   at ROS2.RCLdotnet.Init() in C:\dev\ros2_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet\RCLdotnet.cs:line 316
   at ConsoleApplication.RCLDotnetTalker.Main(String[] args) in C:\dev\ros2_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet_examples\RCLDotnetTalker.cs:line 13`
```
not UWP, just dotnet, but the platform is being detected as UWP?

esteve, 2019-01-07:
Does this work now after https://github.com/esteve/ros2_dotnet/pull/8 was merged?

dvargasfr, 2019-01-08:
Yes, it does. But including the correct version of Fast-RTPS and rcl_interfaces in ros2_dotnet.repos as suggested in #9 .

closed 2019-01-08 by dvargasfr