ros2-dotnet/ros2_dotnet #12: Compiling UWP for HoloLens

jeffdelmerico, 2019-02-04:
Hi Esteve,

I'm trying to compile ros2_dotnet in order to deploy some ros2 applications to HoloLens, so I have been attempting to cross-compile for x86 on my x64 machine.  I am following the instructions in the README, and despite CI saying that it passes for UWP x86, rcldotnet_common fails to build when I select "Visual Studio 15 2017" as the target generator.  The primary error is: 
> error NETSDK1032: The RuntimeIdentifier platform 'win-x86' and the PlatformTarget 'x64' must be compatible

but I've pasted the full output for that package below. Ament builds fine, and I can build successfully for UWP x64.  Any help would be appreciated. Thanks!

```
-- Build files have been written to: C:/workspaces/ros2_dotnet_ws/build/rcldotnet_common ==> 'c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\cmake__build.bat C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\\MSBuild\15.0\bin\msbuild.EXE /m:16 /p:Configuration=Release c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.sln /p:Platform=Win32' in 'c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common'
Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2/4/2019 5:53:22 PM.
-- Detecting C compiler ABI info - done
-- Detecting C compile features
     1>Project "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.sln" on node 1 (default targets).
     1>ValidateSolutionConfiguration:
         Building solution configuration "Release|Win32".
       ValidateProjects:
         The project "INSTALL" is not selected for building in solution configuration "Release|Win32".
         The project "uninstall" is not selected for building in solution configuration "Release|Win32".
     1>Project "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.sln" (1) is building "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\ALL_BUILD.vcxproj.metaproj" (2) on node 1 (default ta
       rgets).
     2>Project "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\ALL_BUILD.vcxproj.metaproj" (2) is building "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\ZERO_CHECK.vcxproj" (4) on node 2 (default targ
       ets).
     4>PrepareForBuild:
         Creating directory "Win32\Release\ZERO_CHECK\".
         Creating directory "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\Release\".
         Creating directory "Win32\Release\ZERO_CHECK\ZERO_CHECK.tlog\".
       InitializeBuildStatus:
         Creating "Win32\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         Checking Build System
         CMake does not need to re-run because C:/workspaces/ros2_dotnet_ws/build/rcldotnet_common/CMakeFiles/generate.stamp is up-to-date.
       FinalizeBuildStatus:
         Deleting file "Win32\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
         Touching "Win32\Release\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
     4>Done Building Project "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\ZERO_CHECK.vcxproj" (default targets).
     1>Project "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.sln" (1) is building "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.vcxproj.metaproj" (3) on node 1 (def
       ault targets).
     3>Project "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.vcxproj.metaproj" (3) is building "c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.vcxproj" (5) on node 1
       (default targets).
     5>PrepareForBuild:
         Creating directory "Win32\Release\rcldotnet_common\".
         Creating directory "Win32\Release\rcldotnet_common\rcldotnet_common.tlog\".
       InitializeBuildStatus:
         Creating "Win32\Release\rcldotnet_common\rcldotnet_common.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/bin/cmake.exe;-version;C:/Program Files/dotnet/dotnet.exe restore;-r;win-x86 rcldotnet_co
         mmon_dotnetcore.csproj; C:/Program Files/dotnet/dotnet.exe publish;-r;win-x86 rcldotnet_common_dotnetcore.csproj -> C:/workspaces/ros2_dotnet_ws/build/rcldotnet_common
         cmake version 3.12.18081601-MSVC_2

         CMake suite maintained and supported by Kitware (kitware.com/cmake).
           Restoring packages for c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common_dotnetcore.csproj...
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
           Generating MSBuild file c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\obj_core\rcldotnet_common_dotnetcore.csproj.nuget.g.props.
           Generating MSBuild file c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\obj_core\rcldotnet_common_dotnetcore.csproj.nuget.g.targets.
           Restore completed in 355.56 ms for c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common_dotnetcore.csproj.
         Microsoft (R) Build Engine version 15.9.20+g88f5fadfbe for .NET Core
         Copyright (C) Microsoft Corporation. All rights reserved.

-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.16.27023/bin/Hostx86/x86/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting C compile features - done
-- Found ament_cmake_python: 0.5.1 (C:/workspaces/ament_ws/install/share/ament_cmake_python/cmake)
[... 39 lines cut ...]
  File "C:\Python37\lib\concurrent\futures\_base.py", line 425, in result
    return self.__get_result()
  File "C:\Python37\lib\concurrent\futures\_base.py", line 384, in __get_result
    raise self._exception
  File "C:\Python37\lib\concurrent\futures\thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "c:\workspaces\ament_ws\install\Lib\site-packages\ament_tools\verbs\build_pkg\cli.py", line 280, in main
    return run(opts, context)
  File "c:\workspaces\ament_ws\install\Lib\site-packages\ament_tools\verbs\build_pkg\cli.py", line 331, in run
    handle_build_action(on_build_ret, context)
  File "c:\workspaces\ament_ws\install\Lib\site-packages\ament_tools\verbs\build_pkg\cli.py", line 270, in handle_build_action
    run_command(build_action, context)
  File "c:\workspaces\ament_ws\install\Lib\site-packages\ament_tools\verbs\build_pkg\cli.py", line 262, in run_command
    sys.exit(msg)
SystemExit: <== Command 'c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\cmake__build.bat C:\Program Files (x86)\Microsoft Visual Studio\2017\Enterprise\\MSBuild\15.0\bin\msbuild.EXE /m:16 /p:Configuration=Release c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common\rcldotnet_common.sln /p:Platform=Win32' failed in 'c:\workspaces\ros2_dotnet_ws\build\rcldotnet_common' with exit code '1'
-- Found rosidl_typesupport_interface: 0.5.1 (C:/workspaces/ros2_dotnet_ws/install/share/rosidl_typesupport_interface/cmake)
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    INSTALL_EXAMPLES
    THIRDPARTY

```

fmrico, 2019-02-04:
Hi @jeffdelmerico ,

May I see your compilation command? I am (just now!!) compiling for UWP 32 (for hololens) in my Windows10 64 bits machine without any problem.

Best

esteve, 2019-02-05:
@jeffdelmerico thanks for trying out ros2-dotnet. It looks like the .NET Core runtime for x86 is not installed in your NuGet cache. The CI uses Azure Pipelines, so the test runners already have everything.

Could you type the following to check that the everything is correct?

```
dotnet --list-runtimes
```

And this to make sure that you can build .NET Core projects for win-x86:

```
dotnet new console -n foo
cd foo
dotnet restore -r win-x86
dotnet build -r win-x86
```

You should now have a file named `foo.exe` in `bin\Debug\netcoreapp2.1\win-x86/`

Thanks.

jeffdelmerico, 2019-02-05:
Hi @esteve and @fmrico 

Thanks so much for developing the dotnet interface for ros2!  

I'm a long time ROS user, but I am new to .NET and Windows development, so it's possible that I've gone off track with my local setup.  However, here are the steps I've taken, starting from a clean terminal with no ROS or ROS2 distributions sourced:

```
md \workspaces\ament_ws\src
cd \workspaces\ament_ws
curl -sk https://raw.githubusercontent.com/esteve/ros2_dotnet/master/ament_dotnet_uwp.repos -o ament_dotnet_uwp.repos
vcs import src < ament_dotnet_uwp.repos
python src\ament\ament_tools\scripts\ament.py build --cmake-args -G "Visual Studio 15 2017 Win64" --
call install\local_setup.bat
cd \workspaces
md ros2_dotnet_ws\src
cd ros2_dotnet_ws
curl -sk https://raw.githubusercontent.com/esteve/ros2_dotnet/master/ros2_dotnet_uwp.repos -o ros2_dotnet_uwp.repos
vcs import src < ros2_dotnet_uwp.repos
ament build --parallel --cmake-args -G "Visual Studio 15 2017" -DCMAKE_SYSTEM_NAME=WindowsStore -DCMAKE_SYSTEM_VERSION=10.0.14393 -DTHIRDPARTY=ON -DINSTALL_EXAMPLES=OFF -DCMAKE_FIND_ROOT_PATH="\workspaces\ament_ws\install;\workspaces\ros2_dotnet_ws\install" --
```

The ament step produces lots of output but no errors.  The ros2_dotnet step produces the error I included initially, along with lots of output for successful package builds.

@esteve: Here is the output of the commands you suggested:
```
# dotnet --list-runtimes
Microsoft.AspNetCore.All 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
Microsoft.AspNetCore.All 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
Microsoft.AspNetCore.All 2.2.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.All]
Microsoft.AspNetCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
Microsoft.AspNetCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
Microsoft.AspNetCore.App 2.2.1 [C:\Program Files\dotnet\shared\Microsoft.AspNetCore.App]
Microsoft.NETCore.App 2.1.6 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
Microsoft.NETCore.App 2.1.7 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]
Microsoft.NETCore.App 2.2.1 [C:\Program Files\dotnet\shared\Microsoft.NETCore.App]

# dotnet new console -n foo
Getting ready...
The template "Console Application" was created successfully.

Processing post-creation actions...
Running 'dotnet restore' on foo\foo.csproj...
  Restoring packages for C:\tmp\foo\foo.csproj...
  Generating MSBuild file C:\tmp\foo\obj\foo.csproj.nuget.g.props.
  Generating MSBuild file C:\tmp\foo\obj\foo.csproj.nuget.g.targets.
  Restore completed in 283.78 ms for C:\tmp\foo\foo.csproj.

Restore succeeded.

# dotnet restore -r win-x86
  Restoring packages for C:\tmp\foo\foo.csproj...
  Installing runtime.win-x86.Microsoft.NETCore.DotNetAppHost 2.2.0.
  Installing runtime.win-x86.Microsoft.NETCore.DotNetHostResolver 2.2.0.
  Installing runtime.win-x86.Microsoft.NETCore.DotNetHostPolicy 2.2.0.
  Installing runtime.win-x86.Microsoft.NETCore.App 2.2.0.
  Restore completed in 9.7 sec for C:\tmp\foo\foo.csproj.

# dotnet build -r win-x86
Microsoft (R) Build Engine version 15.9.20+g88f5fadfbe for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Restoring packages for C:\tmp\foo\foo.csproj...
  Installing Microsoft.NETCore.DotNetAppHost 2.2.1.
  Installing Microsoft.NETCore.DotNetHostResolver 2.2.1.
  Installing Microsoft.NETCore.DotNetHostPolicy 2.2.1.
  Installing Microsoft.NETCore.App 2.2.1.
  Installing runtime.win-x86.Microsoft.NETCore.DotNetAppHost 2.2.1.
  Installing runtime.win-x86.Microsoft.NETCore.DotNetHostResolver 2.2.1.
  Installing runtime.win-x86.Microsoft.NETCore.DotNetHostPolicy 2.2.1.
  Installing runtime.win-x86.Microsoft.NETCore.App 2.2.1.
  Generating MSBuild file C:\tmp\foo\obj\foo.csproj.nuget.g.props.
  Generating MSBuild file C:\tmp\foo\obj\foo.csproj.nuget.g.targets.
  Restore completed in 12.69 sec for C:\tmp\foo\foo.csproj.
C:\Program Files\dotnet\sdk\2.2.103\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.RuntimeIdentifierInference.targets(134,5): error NETSDK1032: The RuntimeIdentifier platform 'win-x86' and the PlatformTarget 'x64' must be compatible. [C:\tmp\foo\foo.csproj]

Build FAILED.

C:\Program Files\dotnet\sdk\2.2.103\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.RuntimeIdentifierInference.targets(134,5): error NETSDK1032: The RuntimeIdentifier platform 'win-x86' and the PlatformTarget 'x64' must be compatible. [C:\tmp\foo\foo.csproj]
    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:13.30
```

I've also verified that I already had the x86 runtime installed (on my system at least).
[image]

So do I understand correctly that this runtime is not being found when I go to build?  How do I make sure that this is in my NuGet cache?

Thanks for your help!

esteve, 2019-02-05:
@jeffdelmerico are you using the "x64 Native Tools Command Prompt for Visual Studio 2017" console when building this, or the "x64_x86 Cross Tools Command Prompt"?

As a workaround you can also unset the PlatformTarget environment variable (`set PlatformTarget=`) and that should work.

I usually just use the normal prompt, `ament` only needs to have CMake and MSBuild in `%PATH%`, but that's it.

jeffdelmerico, 2019-02-05:
@esteve thank you very much for your response.  That was indeed the problem.  I realized last night that I had initially used the x64 command prompt, so I tried again this morning with the x86 one, which also didn't work, but I had forgotten about the cross tools prompt.  The workspace builds now, but I get a few errors of the form:
```
 error MSB3491: Could not write lines to file "visualization_msgs_InteractiveMarkerControl__rosidl_typesupport_introspection_c.dir\Release\visualiz.08EEEE21.tlog\visualization_msgs_InteractiveMarkerControl__rosidl_typesupport_introspection_c.lastbuildstate". The specified path, file name, or both are too long. The fully qualified file name must be less than 260 characters, and the directory name must be less than 248 characters. [C:\workspaces\ros2_dotnet_ws\build\visualization_msgs\visualization_msgs_InteractiveMarkerControl__rosidl_typesupport_introspection_c.vcxproj]
```

However, I'm fairly certain that this is just due to my workspace path being too long, so I will try with \dev or whatever.  I've actually tried to change the 260 character limit for my system according to [these instructions](https://index.ros.org/doc/ros2/Installation/Windows-Development-Setup/#id15), but it did not seem to work, at least in this context.

Thanks for your help!

closed 2019-02-05 by jeffdelmerico

esteve, 2019-02-05:
@jeffdelmerico awesome! I'm glad it helped. Regarding the path length, yeah, unfortunately that's a limitation of Windows, but I think a workaround could be implemented in `ros2-dotnet`. Anyway, let me know if you have any doubts with `ros2-dotnet` for UWP and the HoloLens, I'll be happy to help.

jeffdelmerico, 2019-02-05:
Thanks!  I will definitely be in touch as I move forward, and perhaps I can help contribute to ARViz if there is some overlap in our goals.