ros2-dotnet/ros2_dotnet #14: Initializing node with special character in name

jeffdelmerico, 2019-02-13:
Hi Esteve,

I've encountered some odd behavior when trying to instantiate nodes using a string with a - or whitespace in the name.  I haven't tried with all special characters, but names composed of only alphanumeric characters are fine, and underscores seem to be fine as well.  What happens is that if the string contains a special character (for example the device name obtained from `Windows.Security.ExchangeActiveSyncProvisioning.EasClientDeviceInformation`), the call to create the node appears to succeed (the node object is not null), but neither the node nor any topics published or subscribed are visible to ros2 using `ros2 node list` or `ros2 topic list`.  Obviously the workaround is to just use names that don't have these characters, but it's not clear to me where the problem occurs along the ros2 pipeline, whether it's in the .net client library or the native implementation.  There is no strong urgency for a fix, but other users may find unexpected behavior so I thought it relevant to point it out.

Cheers,
Jeff

esteve, 2019-02-13:
IIRC, nodes whose name contains special characters are not supported in the lower layers of ROS2, so the problem is actually in `ros2-dotnet` for not properly checking the node name before instantiating a node object.

jeffdelmerico, 2019-02-13:
Thanks for your response.  I had not previously encountered this with other languages, and I wasn't able to find anything about naming conventions in the regular ROS2 documentation, but based on your suggestion, I tried something similar in python and got an exception:
```
rclpy.exceptions.InvalidNodeNameException: Invalid node name: node name must not contain characters other than alphanumerics or '_':
  'minimal-publisher'
```

closed 2019-02-13 by jeffdelmerico

esteve, 2019-02-13:
I've opened https://github.com/esteve/ros2_dotnet/issues/15 to address this, but unfortunately I don't know when I'll have time to work on it.

fmrico, 2019-02-17:
Hi @esteve @jeffdelmerico 

I address #15 in #16 

Best