ros2-dotnet/ros2_dotnet #17: Incomplete build of ros2_dotnet compared to release files

MischaRo, 2019-02-21:
Hello @ esteve, I have a question concerning the content, when building ros2_dotnet.
I was building ros2_dotnet_uwp without any errors, but after the building process was finished, I compared my content with the content of your release files. I noticed, that my build had quite a lot of files missing, some of whom I need to work on a project.

Just to make sure: this was my build process and I used the x64_x86 Command Prompt

```
md \dev\ament \src
cd \dev\ament
curl -sk https://raw.githubusercontent.com/esteve/ros2_dotnet/master/ament_dotnet_uwp.repos -o ament_dotnet_uwp.repos
vcs import src < ament_dotnet_uwp.repos
python src\ament\ament_tools\scripts\ament.py build --cmake-args -G "Visual Studio 15 2017 Win64" --
call install\local_setup.bat
cd \dev
md ros2_dotnet_ws\src
cd ros2_dotnet_ws
curl -sk https://raw.githubusercontent.com/esteve/ros2_dotnet/master/ros2_dotnet_uwp.repos -o ros2_dotnet_uwp.repos
vcs import src < ros2_dotnet_uwp.repos
ament build --parallel --cmake-args -G "Visual Studio 15 2017" -DCMAKE_SYSTEM_NAME=WindowsStore -DCMAKE_SYSTEM_VERSION=10.0.14393 -DTHIRDPARTY=ON -DINSTALL_EXAMPLES=OFF -DCMAKE_FIND_ROOT_PATH="\dev\ament \install;\dev\ros2_dotnet_ws\install" --

```
this is a picture of the content of my install/lib folder:

[image]

for example, the folder `rcldotnet` and its content that I am dependant on, is absent from my build

esteve, 2019-02-21:
The files on the release page are no longer updated, I'll remove them soon. In any case, it seems your build didn't complete, could you send me the console output of the `ament` run? Thanks!

MischaRo, 2019-02-26:
Thanks for the feedback.
This is the console output of the ament run:

[image]

Its showing 0 errors, so it looks like it my build did complete?

esteve, 2019-02-26:
@MischaRo Do you happen to have OpenSplice installed? Unfortunately it can't be crosscompiled for UWP, so the `rmw_opensplice_cpp` package fails. You can uninstall OpenSplice and try again, or ignore the `rmw_opensplice_cpp` package by putting a `COLCON_IGNORE` file in the `rmw_opensplice_cpp` directory or by passing `--packages-skip` to `colcon build`.

MischaRo, 2019-02-26:
Hi @esteve 

I just tried to build it again, after you removed the ` rmw_opensplice` from the ` ros2_dotnet_uwp.repos`
This time the build finished without errors and the overall package looks more complete!

esteve, 2019-02-26:
@MischaRo that's great! Can I close this ticket? Have you tried #18 with your newly built workspace?

MischaRo, 2019-02-27:
Hi @esteve,
Yes, I have tried #18 with my build and I got it running!
thanks a lot for your help!

closed 2019-02-27 by MischaRo