ros2-dotnet/ros2_dotnet #18: Using ros2_dotnet_uwp with Unity for Hololens not working

MischaRo, 2019-02-21:
Hello @esteve, I am trying to use ros2_dotnet_uwp with unity for Hololens. I was already discussing the issue with @dvargasfr over here https://github.com/IntelligentRoboticsLabs/ARViz/issues/1
https://github.com/IntelligentRoboticsLabs/ARViz released a manual for using the generated DLLs in a UWP application from Unity. The manual is still present in an older version of the readme (https://github.com/IntelligentRoboticsLabs/ARViz/commit/2c384ee48ad4d3a07994be0f88c8158b439281be)

@dvargasfr gave me this code to try it out. 
I used the files from the ros2-dotnet-uwp-x86.zip release file and a Hololens to test it.

```
I use this script:
using System.Collections;
using System.Collections.Generic;
using UnityEngine;

using ROS2;
using ROS2.Utils;

public class ROS2Talker : MonoBehaviour
{

    private int cont = 0;
    private IPublisher<std_msgs.msg.String> chatter_pub;
    private INode node;

    void Start()
    {
        RCLdotnet.Init();
        node = RCLdotnet.CreateNode("talker");
        chatter_pub = node.CreatePublisher<std_msgs.msg.String>("chatter");
    }

    void Update()
    {
        if (RCLdotnet.Ok())
        {
            std_msgs.msg.String msg = new std_msgs.msg.String();
            msg.Data = "Hello World: " + cont++;
            chatter_pub.Publish(msg);
        }
    }

    public void Button_Click()
    {
        if (RCLdotnet.Ok())
        {
            std_msgs.msg.String msg = new std_msgs.msg.String();
            msg.Data = "Hello World: " + cont++;
            chatter_pub.Publish(msg);
        }
    }
}
```
Unfortunately, after starting the application in Hololens, I get an error massage
`
Exception thrown: 'System.DllNotFoundException' in rcldotnet_common.dll
Exception thrown: 'System.EntryPointNotFoundException' in rcldotnet_common.dll
`

[image]

Can you help me find the cause of this error?

fmrico, 2019-02-26:
Hi @MischaRo ,

During the execution, you can see some exception like those. It is used to detect the platform. It does not mean that it is not working. 

Please, check if there is any node or topic up.

Best

MischaRo, 2019-02-26:
Hi @fmrico, 

thanks fort he update!
I checked again and I have this situation:

After starting the application, I ran `ros2 node list` and the node `/talker` was displayed.
So `node = RCLdotnet.CreateNode("talker");` worked.

After looking at `ros2 topic list` nothing was displayed.
So It seems that ` chatter_pub = node.CreatePublisher<std_msgs.msg.String>("chatter");` did not create a topic “chatter” to publish something.

Is my assumption correct, or did I do something wrong?

fmrico, 2019-02-26:
Hi @MischaRo ,

Only for checking: Are you sure that your router support **wireless** udp multicast? I lost a lot of hours this weekend because udp multicast was working using the ethernet, but it doesn't work using the wireless.

MischaRo, 2019-02-26:
Hi @fmrico,
I think that my router does support wireless udp multicast.
But I also tried running the example above on the Hololens emulator, so I am not sure if the wireless udp multicast is an issue?

MischaRo, 2019-02-27:
Hi @fmrico 
I got it running by using my own build from #17
Thanks a lot for your help!

fmrico, 2019-02-27:
Congrats!!!

Stay tuned for new developments ;)

MischaRo, 2019-03-04:
Hi @frico,
sorry but I have to reopen this issue again.
Up until now, I was running Unty/ROS2 in the Hololens Simulator on my computer and got it working there.

After that, I tried to run it on a real Hololens, but it does not work there.
Every time I run the application, its as you said in the message above: I can see the Node, but it does not show me the Topics.

I tried running wireshark, to see if there are any messages transmitted to my computer.
When I use the emulator, I can see the subscribed messages.

[image]

But when I use the real Hololens I do not receive any RTPS messages.
I have tried this with several routers but it was the same every time.
Are there any other settings I have to keep in mind besides udp multicast?

esteve, 2019-03-04:
@MischaRo have you enabled the Internet capabilities in your app? Do you have minimal example of your project?

esteve, 2019-03-04:
@MischaRo in any case, it'd be good to move this to a new issue, the cause for the original issue was resolved and this is a separate problem. Could you open a new ticket?

MischaRo, 2019-03-07:
@esteve I will open a new issue for this

closed 2019-03-07 by MischaRo