ros2-dotnet/ros2_dotnet #19: Add support for Crystal and Dashing

esteve, 2019-02-28:
Right now `ros2_dotnet` only support Bouncy.

adamdbrw, 2019-06-03:
I will start working on this in a couple of days

esteve, 2019-06-03:
@adamdbrw cool, thanks!

adamdbrw, 2019-06-17:
My effort will be focused on supporting Dashing on Linux. I could use some assistance later with UWP

theseankelly, 2019-07-07:
I'm actively investigating options (including ros2_dotnet) for running ROS2 nodes in UWP apps for HoloLens. Let me know what I can/should do to help, @adamdbrw

esteve, 2019-07-08:
@adamdbrw @theseankelly the main focuses of `ros2_dotnet` are Unity and UWP (precisely for the HoloLens), so while adding support for Dashing on Linux is great, the cool thing about `ros2_dotnet` is that it can run on board the HoloLens and any other UWP device (e.g. Xbox One, Windows IoT, etc.)

adamdbrw, 2019-07-08:
I get it. We can have both though. I would like to push out a branch which @theseankelly can then take and make sure it works with UWP and HoloLens just as well. What do you think?

esteve, 2019-07-08:
@adamdbrw sure, sounds great, submit a pull request and we'll take a look at it. Thanks!

theseankelly, 2019-07-08:
Agreed -- we can definitely have both. I'm guessing the bulk of the work is the same between Mono/UWP so @adamdbrw 's work will be a great starting point. Let me know when it's up and I'll have a look.

adamdbrw, 2019-07-08:
@esteve  @theseankelly 
Things are going well. I have integrated Samuel's improvements and managed to get rcldotnet_examples running. Everything is currently on Crystal and I still need to fix .repo files and test the standalone build (i.e. mess up my Crystal distro paths). 

When I am done, I'll make a pull request which contains the integrated changes from Samuel (Crystal API, nested structures, unit tests etc.) plus some of my fixes.

Next step is porting these to Dashing, which I have a good idea how to do. 

You can expect a pull request this week.

adamdbrw, 2019-07-08:
Starting with Crystal, see #33

esteve, 2019-07-08:
@adamdbrw frankly speaking, I'm not comfortable with that approach. Adding support for Crystal/Dashing should be an evolutionary step (e.g. fix the API, update the CMake macros, etc.), but it seems that #33 replaces `rcldotnet` with Samuel's fork entirely (he should be the one to submit the pull request, though). If his fork works better for you, feel free to use it, no problem at all with that :-)

adamdbrw, 2019-07-08:
@esteve I would argue for you to take another look. This approach saves a lot of work that has already been done. I also was under impression that we agreed to this approach over email.
I don't believe in his or hers, just in what's best. It would be best to have one rclcs project that serves well both for HoloLens guys and Unity guys. 
Your repo, your decision.

esteve, 2019-07-09:
@adamdbrw that wasn't the impression I got from our emails. I thought that Samuel would submit the pull request and I asked for the test suite first, since we could have the CI do the job of ensuring that nothing broke.

I'd say that it'd take me less time to port `ros2-dotnet` to Dashing and add any missing features than reviewing 116 files, of a codebase that I'm not familiar with.

BTW, because Samuel's fork evolved on its own, it contains bugs that were fixed in more recent versions of `ros2-dotnet` (e.g. https://github.com/esteve/ros2_dotnet/pull/33/files#diff-8c049a6c7345d33ff874c1035b83f10bR99 won't detect Windows correctly). This means that I'd have to review code that was already fixed in `ros2-dotnet`, it's surreal.

If the pull request basically replaces `ros2-dotnet` with the fork and you like the latter more, go for it, it's all opensource and there's no pressure to use one or the other.

adamdbrw, 2019-07-09:
@esteve Alright, I can see how this is not what you expected and how it is very difficult to review, especially since you got some fixes going in the meantime. 

The base for my decision was based on my evaluation that the unity_ros2 rclcs code was more evolved structurally and ported to Crystal already. And there were additional features implemented, which I liked. Since my goal is the Dashing port with the features, such approach seemed like a good way to go to reach it. I like avoiding repetitive work. I started with looking at your code for initialization and porting some API, then I wrote a Context class, but later saw that Samuel did that and plenty more. 

So, is there a way you see to get all the goodies into your repo, in a way that you are happy with it? You can propose a constructive plan and I'll decide if it is something I will get done. 

In the worst case, when our goals differ too much, maybe you can just get some of my improvements from my fork when I port to Dashing, to include in your own port (such as maybe the rosidl generator port). I'll be happy to help.

It wouldn't be a problem for me to port either his or your code to Dashing and/or add features, but I like using my time efficiently, I hope you understand.

samiamlabs, 2019-07-10:
Yes @esteve, I did and still do want to submit a pull request attempting to integrate what I have been working on with ros2_dotnet. I have just been to busy with work projects to get around to it.

I'm planning to put some time into bringing .NET/C# support to dashing and don't really care about how or what repo, just that it works and has the features I need. I think would be nice if we could put our efforts into the same project though.

I would be willing to help out with more incremental improvements and dashing migration of the ros2_dotnet wrapper but I suspect that I may need to port my rclcs to dasing first in order to meet deadlines for an upcoming project where I need a bunch of the features in my code. (As breaking it up into incremental changes based on ros2_dotnet to support dashing with all the features will be much slower)

theseankelly, 2019-07-10:
Here's a proposal: Let's get the existing ros2_dotnet project "as-is" updated to dashing (as minimal a port as we can possibly achieve - no new functionality).

Once that's done, @samiamlabs and @adamdbrw 's work will be effectively rebased on top of ros2_dotnet, dashing (whether an actual rebase makes any sense or not is up to them).

I'm also doing this in my spare time, but I have some reasons to put some focused effort into this specific topic over the next couple weeks. Hopefully between @esteve and I we can get ros2_dotnet moved up to dashing and go from there?

adamdbrw, 2019-07-15:
So, I managed to port the message generator to Dashing (some issues to fix, but it's looking good). I want to contribute some changes and will look into how I can make small, manageable pull requests (this will take some time).

theseankelly, 2019-07-15:
I've gotten quite close to a straight Crystal port of ros2_dotnet (with no new features).
- I have a functioning set of .repo files for the complete build, as well as the distributed host/target UWP build
- I have updated/rebased all of @esteve 's compatibility changes to the core ROS2 repos, including a couple new ones (see the repo files for pointers)
- I have ported rcldotnet to use Crystal APIs where needed, including cmake boilerplate for actions (actions are not generated, just like services are not)

Unfortunately I've been hitting a wall over an issue related to rosidl_actions (the python script) not properly converting a .action file into its .msg/.srv files in the file system. The CMake variable generation all looks good -- I believe the python interpreter is failing to find the rosidl_actiosn module. I don't have a good explanation for why it's only in ros2_dotnet, though. Unfortunately I likely won't be able to come back to this until next week.

In the meantime if anyone is interested in working through the rosidl_actions issue, my work so far is staged here:
https://github.com/theseankelly/ros2_dotnet/tree/crystal

esteve, 2019-07-16:
@theseankelly this is awesome, thanks! Feel free to submit a pull request when you think it's ready for merging. Is there anything I can help with?

theseankelly, 2019-07-18:
Hey @esteve -- I've been continuing to look at this. Ran into an issue with how the RMW implementation is selected at runtime through the rmw_implementation library -- clearly this can't be made to work within UWP. My solution so far is to replace 'rmw_implementation' in the CMakeLists.txt and package.xml with a specific implementaton ('rmw_fastrtps_cpp'). However, I see you didn't do this in your repo -- I'm curious if/how you worked around this in UWP for bouncy? Perhaps I'm missing something obvious and don't need to do what I'm doing.

esteve, 2019-07-18:
@theseankelly yeah, `rmw_implementation` uses Poco for loading the appropriate rmw vendor and since UWP won't allow Poco's way of loading arbitrary libraries (LoadLibrary), it won't work. Perhaps Poco could be patched to use LoadPackagedLibrary instead, but that's another can of worms.

Anyway, the trick here is that if `rmw_implementation` only finds one rmw implementation at build time, it'll use that one and skip the Poco stuff. So what I did is to only have the `rmw_fastrtps_cpp` repo in the `colcon` workspace.

Now, `rmw_fastrtps_cpp` got split into static generation (https://github.com/ros2/rmw_fastrtps/tree/master/rmw_fastrtps_cpp) and dynamic (https://github.com/ros2/rmw_fastrtps/tree/master/rmw_fastrtps_dynamic_cpp), so you'll have to ignore one of the two via colcon. The dynamic layer is easier to work with, since it won't require the generator to know that you're targeting a different platform, as it uses DDS XTypes to serialize/deserialize the messages.

esteve, 2019-07-18:
@theseankelly BTW, at some point it'd be good to switch to Cyclone (https://github.com/eclipse-cyclonedds/cyclonedds) when it supports XTypes. FastRTPS' performance is rather poor and the code is not as robust as I wished.

theseankelly, 2019-07-18:
awesome! Thanks for the tip. Had a feeling it would be something more straightforward than what I was doing!

theseankelly, 2020-02-09:
Hey @esteve this can now be closed as of #40

closed 2020-03-08 by esteve