ros2-dotnet/ros2_dotnet #21: Replace ament_tools with colcon

esteve, 2019-02-28:


theseankelly, 2019-07-08:
Just an FYI: It looks like colcon now has proper support for VS generator targets as of: https://github.com/colcon/colcon-cmake/commit/46803e738c86ee406fbc41db58a6f8278d9467ea

I've verified I'm able to build some x86 artifacts for ros_base (Dashing) with `colcon build --mege-install --cmake-args -G "Visual Studio 15 2017" -A Win32`. There are other build issues on x86 of course, but I think we ought to be able to move from ament to colcon with minimal effort when updating to dashing.

esteve, 2019-07-08:
@theseankelly this is excellent, thanks for the heads up. I had to patch `ament_tools` to support Win32 (https://github.com/esteve/ament_tools/commit/c2dce537a5652b80de46ae4b3b44b83538f85a65), but never got around to submit the changes for review, I'll update the instructions and the CI to use `colcon` instead. Thanks!

theseankelly, 2019-07-08:
Sure, to be clear though -- I haven't tested it against ros2_dotnet yet. Just saw that colcon won't force you into x64 anymore, if the -A flag is specified.

adamdbrw, 2019-07-08:
@theseankelly 
If you could adapt .repo file to colcon build for a standalone version (which doesn't require ROS2 installed), that would very much of a help (since this is not my use case). The first pull request for us to start should actually come in an hour or so

PR #34 (2019-07-18): Use colcon instead of ament_tools
This just replaces the use of ament_tools in the README and the CI scripts with colcon

Closes #21
closed 2019-07-18 by esteve