ros2-dotnet/ros2_dotnet #22: wireless udp multicast problem ROS2UWP not working on Hololens

MischaRo, 2019-03-07:
Hello @esteve & @fmrico 
As discussed in #18 I will open this new issue for the upcoming problem.
The situation is as follows:

I have a Unity project where, on start, the following script is executed:

```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;

using ROS2;
using ROS2.Utils;

public class ROS2Talker : MonoBehaviour
{

    private int cont = 0;
    private IPublisher<std_msgs.msg.String> chatter_pub;
    private INode node;

    void Start()
    {
        RCLdotnet.Init();
        node = RCLdotnet.CreateNode("talker");
        chatter_pub = node.CreatePublisher<std_msgs.msg.String>("chatter");
    }

    void Update()
    {
        if (RCLdotnet.Ok())
        {
            std_msgs.msg.String msg = new std_msgs.msg.String();
            msg.Data = "Hello World: " + cont++;
            chatter_pub.Publish(msg);
        }
    }

    public void Button_Click()
    {
        if (RCLdotnet.Ok())
        {
            std_msgs.msg.String msg = new std_msgs.msg.String();
            msg.Data = "Hello World: " + cont++;
            chatter_pub.Publish(msg);
        }
    }
}
```

I have uploaded my minimal example application to:
https://github.com/MischaRo/Ros2UWPminimal

When running the application in the Hololens emulator, everything works as expected. When running ROS2 in the console, I can see the node “takler”, the topic “chatter” and all the messages that are generated

[image]

But running the application on a real Hololens I encountered some problems. Upon reviewing the content in the console, the topic “chatter” was not shown to me.

[image]

Sometimes though, it was displayed, but no data was transmitted over it.

[image]

As discussed in #18 this could be a problem concerning wireless udp multicast. 
I have tried running my example on several different wireless networks, all with the same result.
Are there some other configurations that need to be made in order for the application to work on the Hlolens?

esteve, 2019-03-08:
I'm not sure the assessment that it's caused by a faulty wireless router is entirely correct, my guess is that it's caused by the QoS settings in `rcldotnet` not using the reliable profile, which is not implemented yet unfortunately.

Meanwhile, could you edit the description of the ticket and paste the console output as text? It'll be easier for other people to search it.

jeffdelmerico, 2019-03-08:
I've been running into similar problems to @MischaRo , except that I don't even see the node advertised when running on the HoloLens (while everything works fine in the emulator). 

@esteve Can you please clarify what you mean about the QoS settings?  It looks to me like rcldotnet is already using the [default publisher/subscriber](https://github.com/jeffdelmerico/ros2_dotnet/blob/2b89703636596f5107260e45c8c2811ff35d6897/rcldotnet/rcldotnet_node.c#L39) options from rcl, which in turn use the [default rmw profile](https://github.com/ros2/rcl/blob/97ad0013e23e778bc29575e301e36ac2c1443ff3/rcl/src/rcl/publisher.c#L225), and the default is already set to ["reliable"](https://github.com/ros2/rmw/blob/98e971d17a6fea236762d3145e2b3c219322cf7c/rmw/include/rmw/qos_profiles.h#L47).

esteve, 2019-03-18:
@jeffdelmerico you're right, for a while publishers were created without a specific profile and hence they used best effort links, instead of reliable. Do you have a minimal example of your work? Something that I can use to reproduce the issue? Another issue is that nodes need some time to "warm up" before they are can be discovered by other peers. Could you try waiting a little bit before creating a publisher, and the another bit before publishing messages? Thanks.

jeffdelmerico, 2019-03-18:
@esteve I put together a minimal example that is adapted from the RCLDotnetTalker.cs example.  I set up a Blank UWP App in Visual Studio, and then used the following code for the MainPage.xaml.cs:

```
using System;
using System.Threading.Tasks;
using System.Diagnostics;
using Windows.UI.Xaml.Controls;

using ROS2;

// The Blank Page item template is documented at https://go.microsoft.com/fwlink/?LinkId=402352&clcid=0x409

namespace MinimalCommsExample
{
  /// <summary>
  /// An empty page that can be used on its own or navigated to within a Frame.
  /// </summary>
  public sealed partial class MainPage : Page
  {
    public MainPage()
    {
      this.InitializeComponent();

      Talker();
    }

    private async void Talker()
    {
      RCLdotnet.Init();

      INode node = RCLdotnet.CreateNode("talker");
      await Task.Delay(TimeSpan.FromSeconds(10));

      IPublisher<std_msgs.msg.String> chatter_pub = node.CreatePublisher<std_msgs.msg.String>("chatter");
      await Task.Delay(TimeSpan.FromSeconds(10));

      std_msgs.msg.String msg = new std_msgs.msg.String();

      int i = 1;

      while (RCLdotnet.Ok())
      {
	msg.Data = "Hello World: " + i;
	i++;
	Debug.WriteLine("Publishing: \"" + msg.Data + "\"");
	chatter_pub.Publish(msg);

	// Sleep a little bit between each message
	await Task.Delay(TimeSpan.FromSeconds(1));
      }
    }
  }
}
```
I then followed the instructions [here ](https://github.com/esteve/ros2_dotnet#using-generated-dlls-in-your-uwp-application-from-visual-studio) to link the ros2_dotnet DLLs into the project.

From my desktop, which is connected over the same network, I get no output from `ros2 node list`  (using a Bouncy binary installation of ros2) when I run this app on my HoloLens.  The app appears to run correctly based on the output to the debug console.  When I run the app locally in the HoloLens emulator, I'm able to see 'talker' in the output of `ros2 node list`.  

I have not yet ruled out that my network is blocking device-to-device communication, so please let me know if this works for you.

esteve, 2019-03-18:
@jeffdelmerico thanks for the example. Do you happen to have IPv6 enabled on your local network? I recall having issues with Fast-RTPS and IPv6, but now I can't remember if it was with the ROS2 Android port or UWP.

jeffdelmerico, 2019-03-19:
@esteve Unfortunately I am in a situation where I can't administer my network at all, and IPv6 is enabled, at least partially.  I've been trying to communicate between a HoloLens connected over WiFi to a desktop connected via ethernet to the same network.  Over WiFi only an IPv4 address is assigned, but the desktop receives both IPv4 and IPv6 addresses.  I suspect that the network configuration is responsible, but I will try some more experiments to see if I can isolate the problem.  Does this example work for you?

I should also note that the minimal example app has the `internetClient`, `internetClientServer`, and `privateNetworkClientServer` capabilities enabled in the app manifest.

jeffdelmerico, 2019-03-22:
@esteve I did some more tests, and eventually was able to run this minimal example app on my HoloLens and receive messages on a laptop connected over the same WiFi network.  In principle, our wired network should be seamless with the wireless one, but I have not yet had any success getting messages in or out from the desktop. However, I also found that after my initial success with the HL-to-Laptop test, later on it no longer worked when using the same code, where the node running on the HL was detected but the topics were missing.  I think my problems are just issues with my network and are unrelated to the HoloLens or ros2_dotnet itself, and I will be doing some further experiments to investigate this.

esteve, 2019-03-22:
@jeffdelmerico thanks for the update. Does your desktop PC get an IPv6 address? So does your laptop? In any case, I'm sorry to hear you're having these issues, it should be more straightforward, but debugging it all is a bit difficult unfortunately.

jeffdelmerico, 2019-03-26:
@esteve I think I've found a reasonable workaround.  We have a parallel "guest" network that I was told would not allow device-to-device communication, but it appears that UDP isn't blocked.  So I've been able to do bidirectional communication with both ROS and ROS2 over that guest network, which is fine for now.  Thanks for your help!

esteve, 2019-04-01:
@jeffdelmerico that's great!

@MischaRo how big is your network? Is it IPv6 or IPv4? Is there more than one router? What's the netmask?