ros2-dotnet/ros2_dotnet #24: Problem with setup for UWP ament

ceinem, 2019-03-20:
Hi,
I'm struggeling with the general setup. 
When I run the:
`vcs import src < ament_dotnet_uwp.repos`
command, I get the following error:
`C:\dev\ament>vcs import src < ament_dotnet_uwp.repos
Traceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python37\Scripts\vcs.exe\__main__.py", line 9, in <module>
  File "c:\python37\lib\site-packages\vcstool\commands\vcs.py", line 26, in main
    return entrypoint(args)
  File "c:\python37\lib\site-packages\vcstool\commands\import_.py", line 179, in main
    repos = get_repositories(args.input)
  File "c:\python37\lib\site-packages\vcstool\commands\import_.py", line 59, in get_repositories
    repositories = root['repositories']
KeyError: 'repositories'`

Any advice on what I'm doing wrong?
Thanks!!

esteve, 2019-03-20:
Thanks for trying out `ros2-dotnet`. Could you paste the `ament_dotnet_uwp.repos` file here? Sometimes when fetching it from GitHub, it gets scrambled and therefore `vcs` can' import it.

ceinem, 2019-03-20:
Thanks for the quick reply. 
Sure, here it is. 
[ament_dotnet_uwp.zip](https://github.com/esteve/ros2_dotnet/files/2987908/ament_dotnet_uwp.zip)

(Githup didn't let me upload it like that, so I zipped it)

esteve, 2019-03-20:
@ceinem the .repos file is empty (or more appropriately, it contains the string "404: Not Found"). The contents of the file should be as follows:

```
repositories:
  ament/ament_cmake:
    type: git
    url: https://github.com/ament/ament_cmake.git
    version: 0.5.1
  ament/ament_index:
    type: git
    url: https://github.com/ament/ament_index.git
    version: 0.5.1
  ament/ament_lint:
    type: git
    url: https://github.com/ament/ament_lint.git
    version: 0.5.2
  ament/ament_package:
    type: git
    url: https://github.com/esteve/ament_package.git
    version: windows-escape-values
  ament/ament_tools:
    type: git
    url: https://github.com/esteve/ament_tools.git
    version: win32-platform
  ament/googletest:
    type: git
    url: https://github.com/ament/googletest.git
    version: 4b6e624e78ba3d43c1602ffc80478ee7253e0b04
  ament/osrf_pycommon:
    type: git
    url: https://github.com/osrf/osrf_pycommon.git
    version: 0.1.5
  ament/uncrustify:
    type: git
    url: https://github.com/ament/uncrustify.git
    version: 0.66.1
  ament_dotnet/ament_cmake_export_assemblies:
    type: git
    url: https://github.com/esteve/ament_cmake_export_assemblies.git
    version: master
```

I just realized that the URL for the repos file in the instructions was wrong, I've updated it to master:

https://github.com/esteve/ros2_dotnet/blob/master/README.md#ament

Could you try again? Thanks.

ceinem, 2019-03-20:
It worked! 
Awesome! Thank you!

closed 2019-03-20 by ceinem