ros2-dotnet/ros2_dotnet #27: Nested types for PointCloud

ceinem, 2019-05-02:
Hi esteve, 

thank you so much for the development of ros2_dotnet and for the help on the past few issues that I've posted. 
I'm working on a project with PointCloud2 Messages and as far as I understand it, they require nested messages. So I installed everything from fmricos pull request #10 and it installed without a problem. I created a new Visual Studio project and added all the files and references, as you described in your ReadMe. I additionally added the reference for sensor_msgs_assemblies.dll 
Now I'm able to access the header and some parts of the PointCloud Message, but not the data section of it. 

Any chance you can help me with this? Am I missing some references? Or is the Nested Types Code simply not that far yet. 

Thank you for all the amazing work you're doing on this project!!

fmrico, 2019-05-05:
Did you use my PR for collections?

ceinem, 2019-05-05:
Hi fmrico, 
Thanks for your answer. 
I used your code from the nested-types branch. https://github.com/fmrico/ros2_dotnet/tree/nested_types

I'll try the collections PR this evening. Thanks for that note!

ceinem, 2019-05-06:
Hi fmrico, 
thanks for your help! I just tested it on the HoloLens, works perfectly!!

closed 2019-05-06 by ceinem

fmrico, 2019-05-08:
Excellent!!!