ros2-dotnet/ros2_dotnet #49: Support for Actions

theseankelly, 2020-03-27:


shschaefer, 2022-01-12:
As described in public at ROS World 2021, it was our team's intent to work on this.  Before checking in any code, it would be great to understand if anyone else is working on this, like your team @hoffmann-stefan.

hoffmann-stefan, 2022-01-13:
We are currently not working on this library as we wait which of the open PRs (#87/#84 or #89/#90) get chosen and how they are modified by review. Doing something now would result into huge merge conflicts. (Unless we absolutly need some bugfix or little feature for an internal project) After that we would take another look which features we need, Actions are certainly up on the list. So if no one else would implement them we sooner or later would need and implement them. But If someone else want's to do them first we won't complain. Certanly I would like to do a code review while the PR adding support for actions gets discussed.

But before starting implementing Actions I would like to make sure that the following headlines from #90 are implemented/discussed in the main branch, to avoid touching nearly every other line in the actions code again when doing those afterwards:
- Remove interfaces that expose internal implementation details
- Use SafeHandles for memory managment
- Improve error handling

shschaefer, 2022-01-13:
Thanks @hoffmann-stefan .  The work on Actions was near complete last fall before our overlapping PRs were submitted.  I am simply dusting the code off and completing it.  I have looked to see what the impact of changing Esteve's existing pattern which I tried to follow to what you have done.  Most are trivial with the exception of using SafeHandles.  There is a pile of additional work on error handling that exists in my PR and follow on work I have done to my variant of Services and with the Actions work.  I will see how to fold that in.

I am loathe to remove some of the interfaces as you have done as it creates more difficulty in using mocking frameworks and other interface driven patterns which are common in the C# community.  Though I agree with not exposing internal implementation details, not have a top level interface or wrapper for certain functions can be problematic.

hoffmann-stefan, 2022-01-17:
@shschaefer Ok, thanks for the information and your work on this.

I already thought that the interface change can be controversial. I realy like the arguments made in the book [Framework Design Guidlines](https://www.amazon.de/dp/0135896460/) (.NET specific) about this. Though I know many use mocking frameworks (which need interfaces) to test theire code. I have not used a mocking framework yet, somehow got away with it or didn't have code structured that way that would make it nececary. Im not completly opossed to having interfaces here, but after reading and understanding the chaptor on this my default is not havig interfaces for this kind of library code. Also I sometimes watch the [.NET Api Review](https://apireview.net/about) streams, some time ago there was an intressting discussion on interfaces in the base class library, see video link in this comment: https://github.com/dotnet/runtime/issues/45593#issuecomment-900541842. Though I see we may not have the same constraints as the .NET base class library.

Maybe we should discuss this with an concrete method signitures how we can remove implementation details in interfaces without removing them completly (as compared to the main branch) or how we could add interfaces back in (compared to #90). Like should `Node.CreatePublisher<T>()` return the interface or the abstract base class? Are/will there methods in rcldotnet which would take an interface and how would they get to the implementation details (handle)?

Online snippet of the chapter for interfaces: https://docs.microsoft.com/en-us/dotnet/standard/design-guidelines/abstractions-abstract-types-and-interfaces

shschaefer, 2022-03-01:
@hoffmann-stefan , your observation is spot on that we are quite far away from the constraints of the BCL.  However, it is also true that this is not service code where the IoC pattern has proven its worth and driven many of the interface based test patterns.  I often find that the interface pattern allows me to move quickly, then refactor as the abstractions become clear.

I have completed a first pass of Action support in the ms-iot fork, including minimal introduction of SafeHandles, added basic ROS logging support, etc...  There is a good bit of work to be done to improve error handling and native memory safety.  There is cost and complexity with using SafeHandles that I am working to digest.

I am hopeful that we can get the main branch unstuck. Much of refactoring necessary to move Action support is mechanical.  Would prefer to just get it right.

hoffmann-stefan, 2022-03-23:
@shschaefer, I looked at your current published code in the `dev/actions`
branch, as we are nearing the point where we need actions as well in our team.
I know this is an early prototype, but maybe there are some directions we can
discuss before continuing the implementation.

- https://github.com/ms-iot/ros2_dotnet/tree/dev/actions
- https://github.com/ms-iot/ros2_dotnet/compare/2b0bf8b3ff281693b648b51cb3758c56c978dfe1...ms-iot:b848bd5c3a442d269d00d39fb7b76a46e14f38cf

# Thoughts on the API of ActionClients

I'm not sure about the `ActionClient<TGoalRequest, TGoalResponse, TResultRequest, TResultResponse, TFeedback>`
generic type parameters. I would
rather like something like `ActionClient<TAction, TGoal, TResult, TFeedback>` as
there are only the types that the user defined (in the `*.action` file) that
need to be passed in. The request and response types are an implementation
detail of actions.

So I did an initial try on defining an API surface and doing some early method
definitions in my working directory, but without implementing them, except
sketching a few key methods, but nothing that compiles as a whole yet. But this
gives me enough confidence that this API should be possible to implement much
the same as the services are implemented in #90, but with some additional
tricks. Maybe we can discuss this API proposal down below.

# Porting stuff one direction or the other

> I have completed a first pass of Action support in the ms-iot fork, including minimal introduction of SafeHandles, added basic ROS logging support, etc... There is a good bit of work to be done to [...]

I'm not sure why you are trying to port my changes over to your code, while the
direction in the main branch is not clear yet. The cost and complexity with
`SafeHandles` is already digested over here (#90) and it should be easy to
continue on top of my commits in the same pattern.

What are the things you don't like about my PR that you do that much work to
port some ideas I implemented over there to your branch?

Did I miss something that is in your branch (#84) that is not in my branch?
 - I know of the QoS parameters, but they should be easy to add on top of #90.
 - TF2 is a separate package so this would not need to be merged (maybe only
   some git filter-branch trickery to preserve history of that folder)
 - Support for Hololens shouldn't be touched by my commits either, there are
   only changes in the CI and cmake files?

You did copy a lot of code line by line I did write initially without preserving
my carefully crafted and focused history on implementing one concept after the
other. By doing this the attribution of my code as git commit author gets lost
as well. As you contribute in a different license this may even defacto violate
the license of my contributions, but I'm not a lawyer and I don't know if this
is really the case or not. But I can say that I don't like this fact...

I don't think you did this on purpose or wanted to do something bad, but it may
be a good time to step back and collect the pros and cons how this should be
merged together. Which direction is the easiest to port the changes, where is
less stuff missing, how big is the effort to be done on each side to get all
features? If there is stuff missing can this be ported easily on top of the
other branch as they are completely separate and independent files?

I think most of the additional features should be easier to port on top of my
branch/commits than the other way round, and that would as well preserve my
history for collections and services including the work done on memory
management and error handling. Unless I missed something, see the questions
above.

@esteve: Maybe you have some comments here?

# Planning how work should continue on this

As we need actions in the near future I can do the offer to implement the
proposed API (including changes from the discussion) preferably on top of #90 or
an merged main branch. I still need to ask for the time to implement this, but
[... 302 lines cut ...]
var node = RCLDotnet.CreateNode("test_node");

var actionClient = node.CreateActionClient<Fibonacci, Fibonacci_Goal, Fibonacci_Result, Fibonacci_Feedback>("fibonacci_action");

var goalHandle = await actionClient.SendGoalAsync(
    new Fibonacci_Goal() { Order = 10 },
    (Fibonacci_Feedback feedback) => Console.WriteLine( "Feedback: " + string.Join(", ", feedback.Sequence)));

Fibonacci_Result actionResult = await goalHandle.GetResultAsync();

Console.WriteLine("Result: " + string.Join(", ", actionResult.Sequence));
// or
await goalHandle.CancelGoalAsync();
```

Note: For this example to work there needs to be an `Thread`/`Task` that runs
`RCLDotnet.Spin()`. I have some vague idea how this would work nicely with some kind of
`RCLDotnet.SpinAsync(callback, cancellationToken)` that uses an RCLDotnet specific
[`SynchronizationContext`](https://docs.microsoft.com/en-us/dotnet/api/system.threading.synchronizationcontext?view=net-6.0)
to synchronize all callbacks in the same thread, but this is is may be a separate issue. 

## Action Service

Left open for now, but should use the same overall handling of generic type arguments as client.
Maybe define this after the `ActionClient` API surface is discussed.

shschaefer, 2022-03-23:
Thanks @hoffmann-stefan , it is good to hear that your team needs to use this work.  I have expressed since the first post here, the desire to avoid a repeat of PR#84-90 which I feel we are on course to do again.  If you feel that my current work in a secondary staging branch that I indicated would include your code to resolve mechanical differences and ease upstream merging needs direct attribution to you, I am fine with doing so.  

As you requested, I am happy to step back and figure out a better way.  It looks like the CI issue may be resolved soon and we can merge PR #90 then iterate, branch, etc...  to better collaborate now that you may have time to also work on this.

I have continued the implementation with the goal of getting it to a production-ready state.  There are simply too many unhandled exception paths and other issues all the way down to use of the RCL libraries.  I have resolved the generics issue similar to how you proposed hear to simply have `IActionClient/Server<TAction>`.  I initially attempted to do so without reflection or dynamics, but instead deferred it in order to create a compilable, working implementation that could easily be refactored.  The ActionClient is easier as you have seen in your attempt to sketch an API from mine.  The ActionServer is more complex.

### Porting direction
I have not questioned porting my code on top of PR #90.  I would like to avoid another reimplementation of my work without attribution.  I am directly offering to take your feedback and complete the work to upstream my contribution.

Much of what you have proposed is inline with my working branch - post resolving the generics issue.  As I noted in my branch, circular dependencies in compilation prevented the use of UUID and other base interfaces from within the rcldotnet_common branch.  There were trivial fixes to that available.  My first pass left this intact without disturbing the project structure.

### API Feedback
In regards to library form, I tried to follow the RCLCPP Action implementation pattern instead of the Python pattern.  And then recently to begin to make it more idiomatic C#.  I have made my opinion on that known in feedback to PR #91.

Early on I did not find it desirable to expose a goal handle to ActionClient(s).  The RCL APIs do not separate feedback messages, nor does the RCLCPP impl, for example.  But it is a simple lookup to forward messages tagged with Goal ID.  I wonder what porting an existing node that relies on this would look like?  Later, I found it simplifying for handling status and other callbacks.  Your API definition does not reflect the status callback, but it is trivially added as much of the rest matches what I have implemented modulo naming, SendGoalRequestAsync vs. SendGoalAsync.

I also found no reason to create a shadow enum for goal status.  It is generated as action_msgs.msg.GoalStatus and can be cleanly serialized into and out of its status array.

With the server, I have implemented the core pattern of a node with callbacks as shown in the RCLCPP tutorials.  One each for handling goal requests, goal execution and cancellation.  There is an ActionServerGoalHandle class which wraps impl details within the server.  It is exported to the implementor for use in the execution handler.

### General feedback
When it comes to async and threading, there is a lot more to interaction with the underlying wait set implementation.  Creating an async spin is something that ROS2 has tried to avoid with its use of executors and refactoring towards more general synchronization structures like wait sets and timers.  I ran into many of these issues when attempting to complete a working build.

hoffmann-stefan, 2022-03-29:
@shschaefer: This took some time to write a response, but I hope the length of
this post explains why ;)

> I have expressed since the first post here, the desire to avoid a repeat of PR#84-90 which I feel we are on course to do again. 

For now we should be fine here, I did not start implementing this. Discussing
the API for this feature is the way to avoid this.

Though I have to be honest, depending on when this will be finished (or at least
in a somewhat usable state compatible to the other APIs in #90) we can use this.
On the other hand if this isn't ready when we need the features in our internal
code I may not have a choice but implement this myself in order to avoid
delaying the development of our internal stuff/products.

I can't demand that you finish this in time for us. And I hope you understand me
if I would start implementing this for the reason written above. This is not an
easy situation to be in...

Thats why I made the offer in my last comment that I could implement this
proposed API for actions. This would be *another* way out of this situation
without demanding work from you.

In the case that I implement this for our internal products I would continue
sharing the code in the open and finding a way to contribute this back or
helping integrating your/other implementations into the main branch, including
updating our internal code afterwards to be on the main branch of this project
once actions are merged (with all the qualities described in my first post in
this issue).

> If you feel that my current work in a secondary staging branch that I indicated would include your code to resolve mechanical differences and ease upstream merging needs direct attribution to you, I am fine with doing so.

If this stays a secondary staging branch I don't mind. But for me that was more
work than I would imagine one doing for such a branch if he knew everything
would change pretty much later on. My impression was this is the continuation of
the work in #84 with the intention to be proposed as PR later on.

> I initially attempted to do so without reflection or dynamics, but instead deferred it in order to create a compilable, working implementation that could easily be refactored.

Here we have different styles/priorities of working on such things. You wanted
to get this working somehow in a codebase you know and than
optimizing/refactoring the code to be the one you like. My thoughts are that
this should be doable, let's start right away with implementing the API I wan't
to use as soon as possible, to avoid having to change to much later on.

Not really knowing how you work lead me into believing that this was more than a
test branch.

> The ActionClient is easier as you have seen in your attempt to sketch an API from mine.

I did sketch the API based on `rclpy`, not based on your branch. See technical
stuff below for hints why my API proposal more in line with `rclpy`/`rclcpp`
than your branch.

When I looked at your code I was mainly concerned which APIs/functionality did
you implement, not how they where implemented in detail. There I saw that they
are not in line with what I imagined based on the APIs of `rclpy` and `rclcpp`.
So I made the proposal for the API and addressed some other things as well.

> I have not questioned porting my code on top of PR #90. I would like to avoid another reimplementation of my work without attribution. I am directly offering to take your feedback and complete the work to upstream my contribution.

Thanks, I hope I didn't give to much feedback below in this posts ;) It's a lot
of stuff I address here, I hope I made it as clear as possible with providing my
sources on where I have that information from. I also hope that this is seen as
constructive criticism, it's often hard to argue other opinions on technical
stuff while not sounding rude.

# Technical stuff

## Generic parameters

[... 220 lines cut ...]
    // The cancellation token would "fire" if the executor does shut down for example on `CTRL-C`. So all running async callbacks would stop as soon as possible.

    // some method that uses the builtin HttpClient https://docs.microsoft.com/en-us/dotnet/api/system.net.http.httpclient.getasync?view=net-6.0
    var httpResponse = await GetSomeHttpResponse(cancellationToken);

    Console.WriteLine("This gets executed in the thread that called `RCLDotnet.Spin(...)` because of the SynchronizationContext " +
        "(implicit thread local on the thread that starts the await).");

    return new std.srv.Empty_Response();
}
```

`HttpClient` doesn't know anything about the guard conditions used to implement
this "signal and return to spin thread", this will be abstracted away by the
`SynchronizationContext`. If we didn't provide an `SynchronizationContext` the
part of the method after the await would run in some random background thread,
forcing the user of the API to use additional synchronization primitives like
mutexes or concurrent queues to make this thread safe (This example doesn't
access state, but if it would it would be atomic between await statements in
case of an SingleThreadedExecutor or a CallbackGroup).

`rclpy` supports something like that using
[`Executor.create_task(...)`](https://github.com/ros2/rclpy/blob/8e8c978908f36e4cb5f34b3417eb6f912789f76d/rclpy/rclpy/executors.py#L178-L191),
not sure tough if it supports the python version of the await keyword in that
combination.

hoffmann-stefan, 2022-04-04:
@shschaefer: As written in the last post I did some investigation on how the action server APIs work and compared `rclpy` and `rclcpp`. There are differences like you said before between them, all mainly around how the callbacks are handled. I documented the differences here and provide an C# mapping of both libraries as a starting point for discussion how the API should look like. You started with the callback pattern of the `rclcpp` tutorials, which I also think is the right starting point after looking over this (see below for more words on this).

```C#
public abstract class ActionServerGoalHandle
{
    // Only allow internal subclasses.
    internal ActionServerGoalHandle()
    {
    }

    public abstract Guid GoalId { get; }

    public abstract bool IsActive { get; }

    public abstract bool IsCancelRequested { get; }

    public abstract action_msgs.msg.GoalStatus Status { get; }

    // In `rclpy` this calls the `executeCallback` after setting the state to executing.
    // In `rclcpp` this does not call the `executeCallback` (as there is none) but sets the state for the explicit `AcceptAndDefer` `GoalResponse`.
    public void Execute() => throw null;
    
    internal abstract SafeActionServerGoalHandle Handle { get; }
}

public sealed class ActionServerGoalHandle<TAction, TGoal, TResult, TFeedback> : ActionServerGoalHandle
        where TAction : IRosActionDefinition<TGoal, TResult, TFeedback>
        where TGoal : IRosMessage, new()
        where TResult : IRosMessage, new()
        where TFeedback : IRosMessage, new()
{
    // No public constructor.
    internal ActionServerGoalHandle(
        // ...
    )
    {
    }

    // `rclpy` uses the name `Request`, but the name from `rclcpp` `Goal` fits better.
    public TGoal Goal { get; }

    public override Guid GoalId => throw null;

    public override bool IsActive => throw null;

    public override bool IsCancelRequested => throw null;

    public override action_msgs.msg.GoalStatus Status => throw null;

    public void PublishFeedback(TFeedback feedback) => throw null;

    
    // Decide wich (or both?) of these methods should be exposed.

    // "rclpy style"
    public void Succeed() => throw null;

    public void Abort() => throw null;

    public void Canceled() => throw null;

    // "rclcpp style"
    public void Succeed(TResult result) => throw null;

    public void Abort(TResult result) => throw null;

    public void Canceled(TResult result) => throw null;
}

public enum GoalResponse
[... 266 lines cut ...]
There could/will be multiple versions of the callbacks that could be combined in different combinations. 

I think this is the biggest open question which of the different possible callback types we would provide. Though this should only be expanded later, in the first version start with one, maybe two `CreateActionServer()` variants.

To avoid exploding the overloads it could be beneficial to provide an `ActionServerOptions` type which gets the callbacks in a "builder style".

The differences would be:
- taking the `goalId` vs. not (only for the `goalCallback`)
- `async` vs. not
- taking an `cancellationToken` vs. not
- returning the result as `Task<TResult>` vs calling the method on the `GoalHandle` which takes the `TResult`.
- ...

The simplest would be to start with the `rclcpp` style `acceptCallback`, as this is the lowest level. Then we can investigate how to extend this to some sort of `executeCallback` on top or only make an async version of the `acceptCallback` which is pretty much the same.
As the `rclcpp` style `acceptCallback` doesn't return an value we would implement the `GoalHandle.Succeed(TResult result)` style methods first.

# Future idea: Handling of cancellation

Maybe there is some future possibility to integrate the cancelling of an goal with the .NET handling of `CancellationToken`s. This could make this nicely composable with other async tasks. For example cancel the await for an response of an ROS client or an `Task.Delay()`, so the cleanup code would run faster.

Some thoughts/notes:
- Provide a `acceptedCallback`/`executeCallback` that passes an `CancellationToken` which reflects gets activated when the goal was cancelled.
- Catch an `OperationCancelled` exception from the `executeCallback`/`acceptedCallback` and convert this to an abort of the goal from the server.
  - Open Question: How to define the return value `TResult` in this case?
- Stopping the executor would cancel all "in flight callbacks" as well when provided an `CancellationToken`, but this is something not restricted to actions.

shschaefer, 2022-04-05:
Hi @hoffmann-stefan , as I indicated earlier, I started bottoms up instead of tops down to understand the complexity of the action implementation as opposed to services and topics which are quite straightforward and less intertwined with the asynchronous behaviors of the underlying libraries.  I started with the cpp libraries to keep the problem manageable before defining a desirable API surface.  Through we are not far off when it comes to the API design, my prototyping revealed many things which you are discovering and assures me of what is possible.

Since we are not commenting on my PR, but instead re-hashing the API defined there, I will begin with the most cornerstone API of the `ActionServer`, creation from the Node class.  Contrary to your allowance of null callbacks, I kept these explicit as the semantics of your defaults are not quite correct.  Accepting a cancel request, for example, implies that your code can actually handle a cancellation.  It would be more appropriate to REJECT the cancel request as a default.  It may also seem reasonable to accept and execute a goal when asked, but this directly implies that your server can handle multiple, concurrent goal requests.  I believe that making these choices explicit is better API design.  Your example of the nav2 `SimpleActionServer ` shows the pattern very clearly.  We can easily add nice library or helper classes to make simple actions easy to implement without cluttering the basic API with numerous opaque overloads.

With enumerated values, my preference for idiomatic in structure but not naming shows itself.  Trying to rename and redefine constants that are defined in header files like ACCEPT_AND_EXECUTE adds little value.  Looking at a log coming from code invoked using this library, I would hope that the same constants are used in code.  The addition of default or sentinel values, though semantically complete, are not valid.  They will require effort to keep consistent in behavior and debugging.  If we were to do it, I would prefer to see a name like INVALID_RESPONSE = 0 with `GoalResponse` for example.  Make the fact that they are not valid explicit and in the developer's face.  Maybe we should also give feedback to the ROS maintainers that general lack of default values with enums cause inconsistencies in implementation.

It appears that you are starting to see how the use of different executors makes the asynchronous problem more complicated than just enabling async methods.  We will need to be very explicit about how one would spin asynchronously with a single threaded executor and a seemingly benign wait.  I do want to implement `CancellationToken` more thoroughly and potentially the `SynchronizationContext` pattern, but know that it would not have been effective to get a working implementation to start that way.  I like the idea of having the core callback APIs accept an optional `CancellationToken`.  I also agree that it would be nice to handle cancellation of execution, though I think we will need to do some lower level work to make that consistent.

If you look at my example code in RCLDotNetActionServer.cs, it implements the exact code in your usage example except that it makes the goal and cancel callbacks explicit.  The remainder are different patterns for handling errors and exceptions which should be standardized to clean up the code examples.

I am partial to making the core API an idiomatic C# wrapper of the C++ action pattern.  Additions to be more inline with the Python style seem either out of place or better served in layered helper classes.  Additions to make the code more readable or simple to implement seem righteous, but having many overloads of the same functions to look like multiple API sets does not feel right.  Let's prefer that layered approach.

hoffmann-stefan, 2022-05-09:
> Contrary to your allowance of null callbacks, I kept these explicit as the semantics of your defaults are not quite correct. Accepting a cancel request, for example, implies that your code can [...]

Sorry about that, this was a copy-paste error while translating this from the python code [here](https://github.com/ros2/rclpy/blob/31147813b653b4820c0b7c16b12ee30f8ab29ccb/rclpy/rclpy/action/server.py#L175-L182) (did copy the goal callback to the cancel one without noticing the different return value). I changed this in my post to avoid getting this wrong again later on.

hoffmann-stefan, 2022-06-24:
Some changes I found while implementing this in #94:

- Rename `ActionClient.ServiceIsReady()` to `ActionClient.ServerIsReady()`, move to base class as well.
- Some other small renames in the implementation details for the generated messages.
- The proposed `RosActionSendGoalRequestReflectionCache` does not work as you can't reference the concrete `SendGoalRequest` type for the generic parameter.
  - Found another possibility to get around the cyclic references using properties with the base type `IRosMessage` instead of `UUID` or `Time`
    - see https://github.com/ros2-dotnet/ros2_dotnet/pull/94/commits/37845b06c76620103c44fe6061ee27ebba8030b6
    - This completely avoids reflection to get to the `GoalId`
 and `Stamp` only casts are needed.

closed 2023-08-15 by hoffmann-stefan