ros2-dotnet/ros2_dotnet #51: Improve build instructions for Windows Desktop and Linux

theseankelly, 2020-03-28:
The build instructions for Windows and Linux Desktop aren't ideal -- in particular on these platforms it shouldn't be necessary to compile the entire distribution. Rather, set the project up as an overlay workspace on top of an existing ROS2 installation.

For example, look at how the rust language bindings operate: They only build the core binding package itself, plus any relevant packages containing message IDLs (for message type generation): https://github.com/esteve/ros2_rust

fmrico, 2020-03-30:
Hi @theseankelly 

I have just built in Linux only the next packages in an overlay workspace and the ros2_examples work perfectly:

```
https://github.com/ros2-dotnet/dotnet_cmake_module.git
https://github.com/esteve/ament_cmake_export_assemblies.git
https://github.com/ros2-dotnet/ros2_dotnet.git
https://github.com/ros2/common_interfaces.git 
https://github.com/ros2/rcl_interfaces.git
https://github.com/ros2/rosidl_defaults.git
```

This way of use is a great advance for the adoption of ros2_dotnet :)

guillermo-ruffino, 2020-04-28:
> Hi @theseankelly
>
> I have just built in Linux only the next packages in an overlay workspace and the ros2_examples work perfectly:
>
> ```
> https://github.com/ros2-dotnet/dotnet_cmake_module.git [...]

@fmrico
can you share the vcs export and the commands involved? I'd like to do the same in win

guillermo-ruffino, 2020-04-28:
I just tried @fmrico method and I got this

````
C:\dev\ros2_dotnet>colcon build --merge-install
Starting >>> ament_cmake_export_assemblies
Starting >>> dotnet_cmake_module
Starting >>> rosidl_default_runtime
Finished <<< ament_cmake_export_assemblies [2.91s]
Finished <<< rosidl_default_runtime [3.16s]
Finished <<< dotnet_cmake_module [3.42s]
Starting >>> rcldotnet_common
Finished <<< rcldotnet_common [13.2s]
Starting >>> rosidl_generator_dotnet
Finished <<< rosidl_generator_dotnet [9.28s]
Starting >>> rosidl_default_generators
Finished <<< rosidl_default_generators [3.05s]
Starting >>> builtin_interfaces
Starting >>> std_srvs
Starting >>> lifecycle_msgs
[Processing: builtin_interfaces, lifecycle_msgs, std_srvs]
Finished <<< std_srvs [35.4s]
Finished <<< builtin_interfaces [40.9s]
Starting >>> std_msgs
Starting >>> action_msgs
Starting >>> rcl_interfaces
Starting >>> rosgraph_msgs
Finished <<< lifecycle_msgs [55.0s]
[Processing: action_msgs, rcl_interfaces, rosgraph_msgs, std_msgs]
Failed   <<< action_msgs        [ Exited with code 1 ]
Aborted  <<< rosgraph_msgs
Aborted  <<< rcl_interfaces
Aborted  <<< std_msgs

Summary: 9 packages finished [2min 37s]
  1 package failed: action_msgs
  3 packages aborted: rcl_interfaces rosgraph_msgs std_msgs
  14 packages not processed
````

I made myself a repos file based on the list above

````yaml
repositories:
  ros2_dotnet/dotnet_cmake_module:
    type: git
    url: https://github.com/ros2-dotnet/dotnet_cmake_module.git
    version: master
  ament_dotnet/ament_cmake_export_assemblies:
    type: git
    url: https://github.com/esteve/ament_cmake_export_assemblies.git
    version: master
  ros2_dotnet/ros2_dotnet:
    type: git
    url: https://github.com/ros2-dotnet/ros2_dotnet.git
    version: master
  ros2/common_interfaces:
    type: git
    url: https://github.com/ros2/common_interfaces.git
    version: eloquent
  ros2/rcl_interfaces:
    type: git
    url: https://github.com/ros2/rcl_interfaces.git
    version: eloquent
  ros2/rosidl_defaults:
    type: git
    url: https://github.com/ros2/rosidl_defaults.git
    version: eloquent
````

Error is not self descriptive unfortunately

theseankelly, 2020-04-28:
I was working on this exact topic on Windows this morning and I see (I'm pretty sure) the same error as you. It looks like the dotnet/ament cmake scripts are failing to either export or import the assemblies for nested messages, which causes action_msgs to fail. I have no idea (yet) why this problem doesn't exist when building the entire ROS2 source at once.

Hopefully it's a quick fix, but I ran out of time today. I'm going to pick it back up tomorrow morning (I work on this before my day job) and see if I can resolve the issue.

In the meantime if it's urgent, you can consider rolling the ros2_dotnet repository back to the commit just before the nested types change came in: f928c55e81a39431900fc7c6dd21310e2f062a74

guillermo-ruffino, 2020-04-28:
Alright, I'll try the rollback, may be I learn more if I can start from something working.

As you'd might know already, the issue is finding the type `unique_identifier_msgs.msg.UUID` for `Goal_id`

By reading your post I had the idea to compare the generated project (I realized this is generated because it's not well formatted and also on build folder, please don't hesitate on teaching me obvious things I'm very new here)

I found out that when using the underlay this library is not referenced in the generated project:

````xml 
    <Reference Include="C:\dev\ros2_dotnet_ws\install\lib\unique_identifier_msgs\dotnet\unique_identifier_msgs_assemblies.dll" />
````

theseankelly, 2020-04-28:
Yup, that's exactly the missing reference I was running into morning. Will dig in more later!

theseankelly, 2020-04-29:
Well, it was indeed an easy fix. The `unique_identifier_msgs` assembly reference is missing because we are not building `unique_identifier_msgs`! I thought I was, but I had cloned it to the wrong directory by mistake.

If you add https://github.com/ros2/unique_identifier_msgs to your repos list (eloquent branch of course) this should work fine.

I'm currently encountering a Unicode error in `test_msgs` which I don't fully understand so you might need to add `--packages-ignore test_msgs` to your build command.

guillermo-ruffino, 2020-04-29:
Thanks! It worked. Now I don't have the `test_msgs` on my local copy

PR #61 (2020-05-18): Build as an overlay on Windows/Linux
- Updated repos file to build as an overlay
- Upated build instructions
- Created a Linux build workflow

Closes #51 
Closes #53 
Closes #55
commit fce92096 (2020-05-18): Build as an overlay on Windows/Linux

- Updated repos file to build as an overlay
- Upated build instructions
- Created a Linux build workflow

Closes #51 #53 #55
closed 2020-05-18 by esteve