ros2-dotnet/ros2_dotnet #52: Integrate CycloneDDS

esteve, 2020-03-29:
[CycloneDDS](https://github.com/eclipse-cyclonedds/cyclonedds) has become the default DDS implementation for many projects and organizations and ADLINK is actively supporting it for ROS 2.

CycloneDDS is small, fast and has few dependencies, so it'd be ideal for integrating it into UWP.

esteve, 2020-03-29:
One major reason for investigating the use of CycloneDDS in ros2-dotnet is that  Fast-RTPS breaks the UWP builds in every release, despite claiming to have support for it, the eProsima folks don't seem interested in addressing these issues and add UWP to their CI.

fmrico, 2020-03-29:
I agree with @esteve. Besides, results (delay, stability and data loss) with Cyclone are better than Fast-RTPS in our application and sounds reasonable to move to Cyclone.

theseankelly, 2020-03-29:
Fully agree. I took a run at a UWP compatible Cyclone build a couple weeks ago -- there's a large dependency on legacy Win32 APIs which are not allowed in UWP. It needs quite a bit of work unfortunately.

For non-UWP applications, should be no problem to move to Cyclone right now as there's no direct dependency between ros2_dotnet and the DDS implementation.

theseankelly, 2020-03-29:
Also, as of Eloquent I no longer see a straightforward path for ARM64 (HoloLens2) support on Fast-RTPS. This fact increases the need to get Cyclone working -- it's general lack of third party dependencies is attractive for being able to maintain UWP and cross-arch support.

theseankelly, 2020-03-29:
Hey @esteve I see you assigned yourself to this task -- are you actively working on it and/or interested in doing so?

esteve, 2020-03-29:
@theseankelly yeah, I want to give it a try. Do you have any feedback when you attempted to compile CycloneDDS for UWP?

theseankelly, 2020-03-29:
@esteve I didn't try for very long at all so not much to report. Couple things come to mind:
1. Cyclone's build scripts use CMAKE_SYSTEM_NAME to choose between windows/mac/linux. Since this value is set to WindowsStore for UWP (which does not match Windows) it falls through and tries to build the linux targets. You'll need to handle that - maybe here:

https://github.com/eclipse-cyclonedds/cyclonedds/blob/6413d71599aec646eaa427655a0b48bdfaeb4393/src/ddsrt/CMakeLists.txt#L66

From there, its standard practice of finding and replacing all the restricted APIs that MSVC flags when building for UWP. I saw a lot of network interface related stuff, and didn't actually take it further at the time, so no advice there.