ros2-dotnet/ros2_dotnet #53: Problems building for Win10-Desktop

fmrico, 2020-03-29:
Hi,

I am building in Windows10 for Desktop, and I have found some problems:

- `ros2lifecycle` has a test dependency with `ros2lifecycle_test_fixtures`, and it depends on `rclcpp` and `rclcpp_lifecycle`. As @theseankelly and I discussed, it's better to avoid including `rclcpp`.
- `rcl_logging_spdlog` depends on `splog_vendor`, but it is not included in repos.
- `ros2trace` fails for a `UnicodeDecodeError:`

The build is successful if I compile with this line:

```
colcon build --merge-install --packages-ignore ros2lifecycle_test_fixtures rcl_logging_spdlog  ros2trace --cmake-args -DBUILD_TESTING=OFF -DRCL_LOGGING_IMPLEMENTATION=rcl_logging_noop
```

Building commands in CI already include some of these options, but in the instructions in README.md for building Desktop only says to build with `colcon build --merge-install`

@theseankelly Do you think that we should change the build instructions in readme.md? In any case, I don't know if it is recommendable to go without tests...

theseankelly, 2020-03-29:
Yes we should update the instructions -- I have a similar build command as you in the github actions file (https://github.com/ros2-dotnet/ros2_dotnet/blob/f928c55e81a39431900fc7c6dd21310e2f062a74/.github/workflows/build_desktop.yml#L47) but forgot to update the README!

Nonetheless, per my comments in #51 I think it would be better (for Linux and Desktop) to not require an entire rebuild. It should be possible to build as an overlay from an existing ROS2 install -- that way all you would need is the ros2_dotnet projects plus any packages containing messages you want .NET bindings generated for. How does that sound?

fmrico, 2020-03-29:
>  It should be possible to build as an overlay from an existing ROS2 install -- that way all you would need is the ros2_dotnet projects plus any packages containing messages you want .NET bindings generated for. How does that sound?

This sounds great!!!

mce16, 2020-04-13:
i run from command prompt:
colcon build --merge-install

Turn out some errors encountered, anyone has such problem?

Starting >>> ament_package
Starting >>> ament_lint
Starting >>> ament_cppcheck
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> gtest_vendor
Starting >>> tinydir_vendor
Starting >>> osrf_pycommon
Fatal Python error: failed to get random numbers to initialize Pythont - 0.4s .....

mce16, 2020-04-15:
Finally managed to build, but encounter some errors as below; would appreciate any pointer;

`
Starting >>> ament_package
Starting >>> ament_lint
Starting >>> ament_cppcheck
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> gtest_vendor
Starting >>> tinydir_vendor
Starting >>> osrf_pycommon
Finished <<< ament_cppcheck [4.48s]
Starting >>> ament_pep8
Finished <<< ament_package [5.39s]
Finished <<< ament_lint [5.44s]
Starting >>> ament_cmake_core
Starting >>> ament_flake8
Finished <<< osrf_pycommon [7.08s]
Finished <<< ament_pep8 [4.39s]
Starting >>> ament_pyflakes
Finished <<< gtest_vendor [9.25s]
Starting >>> gmock_vendor
Finished <<< ament_flake8 [4.80s]
Starting >>> ament_pep257
Starting >>> ament_mypy
Finished <<< ament_cmake_core [7.27s]
Finished <<< ament_pyflakes [3.92s]
Starting >>> ament_cmake_libraries
Starting >>> ament_cmake_export_libraries
Finished <<< gmock_vendor [5.05s]
Starting >>> ament_cmake_include_directories
Finished <<< ament_pep257 [4.38s]
Starting >>> ament_copyright
Finished <<< ament_mypy [4.94s]
Starting >>> ament_cmake_python
Finished <<< ament_cmake_libraries [8.17s]
Finished <<< ament_cmake_export_libraries [8.23s]
Starting >>> ament_cmake_export_dependencies
Starting >>> ament_cmake_export_definitions
Finished <<< ament_copyright [7.49s]
Starting >>> ament_cmake_export_include_directories
Finished <<< ament_cmake_include_directories [7.89s]
Starting >>> ament_cmake_export_link_flags
Finished <<< ament_cmake_python [7.38s]
Starting >>> ament_cmake_test
Finished <<< fastcdr [22.9s]
Starting >>> ament_cmake_version
Finished <<< ament_cmake_export_definitions [6.23s]
Starting >>> ament_cmake_export_interfaces
Finished <<< ament_cmake_export_dependencies [7.16s]
Starting >>> ament_cmake_target_dependencies
Finished <<< ament_cmake_export_include_directories [6.41s]
Finished <<< ament_cmake_export_link_flags [6.34s]
Starting >>> ament_xmllint
Starting >>> ament_lint_cmake
Finished <<< ament_cmake_version [6.03s]
Starting >>> ament_index_python
Finished <<< tinydir_vendor [29.2s]
Starting >>> ament_cpplint
Finished <<< ament_cmake_test [6.84s]
Starting >>> ament_lint_auto
Finished <<< ament_xmllint [6.34s]
Finished <<< ament_lint_cmake [6.38s]
Finished <<< ament_cmake_target_dependencies [6.92s]
Finished <<< ament_cmake_export_interfaces [7.81s]
Starting >>> ament_cmake
Starting >>> ament_cmake_lint_cmake
Starting >>> ament_cmake_gtest
Starting >>> ament_cmake_pytest
Finished <<< ament_index_python [6.72s]
[... 76 lines cut ...]
[6.444s] colcon.colcon_ros.task.ament_python.build WARNING Package 'osrf_pycommon' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                                
[10.353s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_pep8' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                                  
[11.254s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_flake8' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                                
[14.234s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_pyflakes' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                              
[16.090s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_pep257' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                                
[16.730s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_mypy' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                                  
[20.370s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_copyright' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                             
[34.444s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_xmllint' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                               
[35.090s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_lint_cmake' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                            
[35.661s] colcon.colcon_ros.task.ament_python.build WARNING Package 'ament_index_python' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                          
[... cut ...]
[130.622s] colcon.colcon_ros.task.ament_python.build WARNING Package 'launch_yaml' doesn't explicitly install the 'package.xml' file (colcon-ros currently does it implicitly but that fallback will be removed in the future)                                                                
Failed   <<< uncrustify_vendor  [ Exited with code 1 ]                                                                                                                                                                                                                                        
D:\ros2_dotnet_ws>        
`

mce16, 2020-04-15:
Hi, managed to run the build, but not sure why i am encountering this error  
rcldotnet_common   [ Exited with code 1 ]           
towards the end of the build; and the rest of the rcldotnet_examples, rosidl_generator_dotnet all are not built.

`

Starting >>> rcl_yaml_param_parser                                                                                                                                                                        
Failed   <<< rcldotnet_common   [ Exited with code 1 ]                                                                                                                                                    
Aborted  <<< rosidl_generator_dds_idl                                                                                                                                                                     
Aborted  <<< rcpputils                                                                                                                                                                                    
Aborted  <<< rcl_logging_spdlog                                                                                                                                                                           
Aborted  <<< rcl_yaml_param_parser                                                                                                                                                                        
Aborted  <<< rcl_logging_log4cxx                                                                                                                                                                          
Aborted  <<< rcl_logging_noop                                                                                                                                                                             
Aborted  <<< rosidl_generator_c                                                                                                                                                                                                                                                                                                                                                                                     
Summary: 87 packages finished [2min 10s]                                                                                                                                                                    
1 package failed: rcldotnet_common                                                                                                                                                                        
7 packages aborted: rcl_logging_log4cxx rcl_logging_noop rcl_logging_spdlog rcl_yaml_param_parser rcpputils rosidl_generator_c rosidl_generator_dds_idl                                                   
60 packages not processed
`

PR #61 (2020-05-18): Build as an overlay on Windows/Linux
- Updated repos file to build as an overlay
- Upated build instructions
- Created a Linux build workflow

Closes #51 
Closes #53 
Closes #55
closed 2020-05-18 by esteve