ros2-dotnet/ros2_dotnet #55: Build error on Windows 10 dekstop

mce16, 2020-04-13:
i run from command prompt:
colcon build --merge-install

Turn out some errors encountered, anyone has such problem?

Starting >>> ament_package
Starting >>> ament_lint
Starting >>> ament_cppcheck
Starting >>> fastcdr
Starting >>> poco_vendor
Starting >>> gtest_vendor
Starting >>> tinydir_vendor
Starting >>> osrf_pycommon
Fatal Python error: failed to get random numbers to initialize Pythont - 0.4s .....

comoc, 2020-04-28:
same for me.

```
Starting >>> rcl_logging_log4cxx
Starting >>> rcl_logging_noop
Starting >>> rcl_logging_spdlog
Starting >>> rcl_yaml_param_parser
--- stderr: rcldotnet_common
CMake Error at C:/dev/ros2_dotnet_ws/install/share/dotnet_cmake_module/cmake/Modules/FindDotNETExtra.cmake:16 (include):
  include called with wrong number of arguments.  include() only takes one
  file.
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

CMake Error at C:/dev/ros2_dotnet_ws/install/share/dotnet_cmake_module/cmake/Modules/FindDotNETExtra.cmake:26 (csharp_add_project):
  Unknown CMake command "csharp_add_project".
Call Stack (most recent call first):
  CMakeLists.txt:17 (add_dotnet_library)

---
Failed   <<< rcldotnet_common   [ Exited with code 1 ]
Aborted  <<< rosidl_generator_dds_idl
Aborted  <<< rosidl_generator_c
Aborted  <<< rcpputils
Aborted  <<< rcl_logging_spdlog
Aborted  <<< rcl_yaml_param_parser
Aborted  <<< rcl_logging_log4cxx
Aborted  <<< rcl_logging_noop

Summary: 87 packages finished [1min 53s]
  1 package failed: rcldotnet_common
  7 packages aborted: rcl_logging_log4cxx rcl_logging_noop rcl_logging_spdlog rcl_yaml_param_parser rcpputils rosidl_generator_c rosidl_generator_dds_idl
  1 package had stderr output: rcldotnet_common
  60 packages not processed
```

theseankelly, 2020-04-28:
I believe this is a duplicate of #53 -- have you tried the workaround posted by @fmrico?

In the meantime, I'm working on an improved set of build instructions for Windows/Linux. There ought to be no need to build the entire ROS2 source unless you're trying to build for UWP.

quattrinili, 2020-05-16:
I also the same problem and the workaround in #53 does not work in solving the problem -- I believe that workaround applies for the Windows Desktop instruction and not the UWP. If I have time I will dig a little more into it.

quattrinili, 2020-05-16:
At least the rcldotnet_common error was solved for the UWP build -- it was an incompatibility with the platform. (To get the specific error, add to the colcon command the option
`--event-handlers console_cohesion+`; in this way the errors printed in the standard output are also displayed.)

The next error now is on fastrtps and I suspect a difference with the language being used, given that it gives some syntax errors. 

e.g., error C2332: 'struct': missing tag name (compiling source file ros2-uwp\src\eProsima\Fast-RTPS\src\cpp\transport\TCPAcceptorSecure.cpp) 

I will keep digging and see what I will find.

quattrinili, 2020-05-16:
At least for me, the workaround to make the UWP build work was to remove the following flag 
`-DCMAKE_SYSTEM_NAME=WindowsStore` resolving the error about fastrtps in the previous comment.

theseankelly, 2020-05-17:
Hey @quattrinili -- the `-DCMAKE_SYSTEM_NAME=WindowsStore` flag is required for UWP. The build will work but you'll likely find runtime problems in UWP.

Can you share your full `colcon build ...` command you are using for UWP?

theseankelly, 2020-05-17:
On second thought @quattrinili if this is for UWP can you please open a new issue?

esteve, 2020-05-17:
@quattrinili 

Thanks for your report, though it'd better to file a separate ticket, given that the issue you encountered differs from this ticket.

> The next error now is on fastrtps and I suspect a difference with the language being used, given that it gives some syntax errors.
> 
> e.g., error C2332: 'struct': missing tag name (compiling source file ros2-uwp\src\eProsima\Fast-RTPS\src\cpp\transport\TCPAcceptorSecure.cpp)

FastRTPS keeps breaking the UWP support in every release, despite @theseankelly and I having sent patches to fix it, but they always revert the changes in new releases and don't even care about regression tests.

Just to make sure this is not a new issue in FastRTPS, can you check the following?

- Are you using our fork of FastRTPS? https://github.com/ros2-dotnet/Fast-RTPS

- Are you running this with the repos file for UWP? https://github.com/ros2-dotnet/ros2_dotnet/blob/master/ros2_dotnet_uwp.repos

Thanks!

quattrinili, 2020-05-17:
Thank you @theseankelly and @esteve. Indeed the DCMAKE_SYSTEM_NAME=WindowsStore flag is required for UWP -- I just tested to see if it was working, but it was throwing an exception in loading the DLL in an UWP app. The error looked similar at the  beginning, thus I commented here, but it is indeed a different issue, so I opened the new issue, responding to your questions as well. I will continue in the other issue for other new findings.

PR #61 (2020-05-18): Build as an overlay on Windows/Linux
- Updated repos file to build as an overlay
- Upated build instructions
- Created a Linux build workflow

Closes #51 
Closes #53 
Closes #55
closed 2020-05-18 by esteve