ros2-dotnet/ros2_dotnet #56: Consider using new `rclc` client library

theseankelly, 2020-04-27:
The Micro-ROS folks have been working on a c-language abstraction on top of rclc.

We might consider building ros2_dotnet on top of this layer moving forward rather than re-implementing clients/services/actions directly on top of rcl.

At the moment `rclc` only supports Foxy. I'll have a closer look at the project and see if it's viable for ros2_dotnet.

theseankelly, 2020-05-09:
Hey @esteve I'd love your thoughts on this topic. I imagine you started this project with `rcl` because the `rclpy` implementation provided a good model. I'm curious how you feel about that decision 3-4 years later as things have matured -- any comments on the summary above? What am I not thinking about?

fmrico, 2020-05-09:
I would love to, but maybe we are breaking the ROS2 design. It's supposed that any client library is on top on rcl... I want to know the opinion of @esteve

theseankelly, 2020-05-09:
That's a good point. I suppose such a choice would turn this project into *just* language bindings for another client library instead of a full blown standalone client library. I'm not yet sure if that's a good or a bad thing.

esteve, 2020-05-09:
> Hey @esteve I'd love your thoughts on this topic. I imagine you started this project with rcl because the rclpy implementation provided a good model. I'm curious how you feel about that decision 3-4 [...]

The rationale for `rcl` is that each client library may have its own language-specific peculiarities and you want those client libraries to "feel" natural to the developers. One example is the threading model, it's very language-specific, so you don't want to use something that may feel alien (e.g. using `pthread` from .NET may be awkward) and that is difficult to integrate with constructs provided by the language (e.g. `await`, `async`, etc. in C#)

Nothing stops your from wrapping `rclcpp` of course, and nothing is set on stone, but it's a matter of making ROS 2 client libraries feel natural instead of forcing other language-specific constructs, at the expense of requiring more work from client library developers. Moreover, it also helps accepting contributions, since some contributors may not feel comfortable with C or C++, but are used to the target language.

theseankelly, 2020-05-09:
Great points. I suppose replicating some implementation logic (executors, services, etc) in the target language would also make debugging easier for users. I was coming at this with my C/C++ hat on, not thinking about it from the perspective of .NET users.

Makes sense for me, I'll call this topic closed for now.

fmrico, 2020-05-09:
I will play the role of "agent provocateur". Are we willing to develop in C# what is currently working in C++? Maybe not, but I think using rcl is the correct decision

theseankelly, 2020-05-09:
[I closed prematurely, we can keep discussion ongoing for a while ;)]

Agree, it's a tough choice between what's 'right' and what gets the job done most efficiently. For my purposes (ROS in UWP containers on HoloLens) I do feel quite torn about the best path forward: pushing ros2_dotnet forward and fleshing out the implementation versus writing UWP component libraries in C++/WinRT, including custom APIs to project into UWP that way.

I do of course think ros2_dotnet has merit on its own right, as it's usage extends beyond just UWP and even Windows.

fmrico, 2020-05-09:
IMHO the goal is to make ros2_dotnet attractive enough to create a community around it. Maybe then C# developers could start to improve the core, and then the decision is clearly to build it on top of rcl as an independent client library. That would be very cool. There are many SDKs only in C#. This project is very useful.

Moving ros2_dotnet to eloquent has been a great improvement and maybe the only way to let this project to survive. The building, as an overlay workspace is a big step. We should push to get the basic functionality working: collections (I am on it), services, actions, at least

I am very interested in what you said about Hololens. We worked the project ARViz for hololens, but I think that the result was not the best. Now we have another project MOCAP4ROS2 in which we are using ros2_dotnet because the SDK of our IMUs is in C#

esteve, 2020-05-09:
> Agree, it's a tough choice between what's 'right' and what gets the job done most efficiently. For my purposes (ROS in UWP containers on HoloLens) I do feel quite torn about the best path forward: pushing ros2_dotnet forward and fleshing out the implementation versus writing my ROS nodes as UWP component libraries in C++/WinRT, including custom APIs to project into UWP that way.

I'd say the scope of `ros2_dotnet` is CLR languages, more specifically languages supported by .NET Core. So, for developers who want to write C++/WinRT applications, it's just best to ensure that upstream `rclcpp` can target WinRT. However, for those who want to use C#, F#, etc. `ros2_dotnet` is useful by itself, plus it'd be more attractive to them in case they want to contribute so they don't have to be comfortable with C++.