ros2-dotnet/ros2_dotnet #59: Exception thrown: 'System.DllNotFoundException' in rcldotnet_common.dll

mce16, 2020-05-05:
Hi

I managed to run the talker and listener example using the binary built.

However, when i try to use the dlls in my Visual Studio 2017 project, although the talker and listener everything seems to run smoothly; i encountered the following in the output:
"Exception thrown: 'System.DllNotFoundException' in rcldotnet_common.dll"

I saw this issue reported by another user before; but the user has actually having another issue at the same time. So i believe not quite related to his. I am just wondering if i should bother about it; if this is just a normal output?  because it is sending the output every time the ROS2 codes or function is called.

theseankelly, 2020-05-05:
Can you use loader snaps to determine the missing dll?

https://docs.microsoft.com/en-us/archive/blogs/junfeng/debugging-loadlibrary-failures

mce16, 2020-05-07:
Thanks.

I have used the gflags, here is the output from windbg; (the gflags somehow doesn't work with my visual studio debugger, its output doen't show anything, not sure why).

WindowsFormApp1.exe  has made a call to RCLdotnet.Init() during initialization.

[log.TXT](https://github.com/ros2-dotnet/ros2_dotnet/files/4590590/log.TXT)

theseankelly, 2020-05-08:
It doesn't look like any DLL has actually failed to load as of that breakpoint. Let the debugger go and see if it progresses to give any errors about dll loading in the text output.

mce16, 2020-05-10:
Sorry, i think i have attached the wrong log file, it should be this:

[log2.TXT](https://github.com/ros2-dotnet/ros2_dotnet/files/4605246/log2.TXT)