ros2-dotnet/ros2_dotnet #60: Build process does not progress

comoc, 2020-05-06:
After ```colcon build --merge-instal```, do not proceed as shown below.

```
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[Processing: uncrustify_vendor]
[7min 45.2s] [68/155 complete] [uncrustify_vendor:build - 7min 12.2s]
```

The patch.exe window remains visible during this display.

* Windows 10 Pro
* Visual Studio 2019
* patch.exe was installed via Chocolatey.

Thanks.

theseankelly, 2020-05-06:
Doesn't sound like a ros2_dotnet specific issue.

Sounds like what happens when you don't run your `colcon build --merge-install` command in an Administrative command prompt -- patch elevates, opens new cmd window, and nothing continues. Possible?

comoc, 2020-05-06:
Hi @theseankelly ,
Thank you for the reply.

I had tried it on non administrative command prompt before.
So I have tried on administrative command prompt again, then the patch process looked successful.

And the result turned the following.
```
Summary: 52 packages finished [36.3s]
  1 package failed: uncrustify_vendor
  11 packages aborted: ament_cmake_auto ament_cmake_clang_format ament_cmake_clang_tidy ament_cmake_cppcheck ament_cmake_cpplint ament_cmake_flake8 ament_cmake_mypy ament_cmake_pclint ament_cmake_pep257 ament_cmake_xmllint libyaml_vendor
  91 packages not processed
```

What should I do next?

Thanks,

esteve, 2020-05-20:
@comoc can you confirm these details so we can help you better:

- Have you installed all the dependencies as detailed in the README?
- What operating system are you building `ros2_dotnet`?
- If you're building on Windows, is it for UWP or Desktop?
- What version of ROS 2 is installed on your system?

Thanks!

comoc, 2020-05-22:
Hi @esteve,

I tried the latest version of ros2_dotnet. Finally, It worked!!!

> If you're building on Windows, is it for UWP or Desktop?
Desktop

> What version of ROS 2 is installed on your system?
https://github.com/ros2/ros2/releases/download/release-eloquent-20200124/ros2-eloquent-20200124-windows-release-amd64.zip

After that, I faced the following problem.
```
C:\dev\ros2_dotnet_ws>call \dev\ros2_dotnet_ws\install\local_setup.bat

C:\dev\ros2_dotnet_ws>ros2 run rcldotnet_examples rcldotnet_talker
Multiple executables found:
- C:\dev\ros2_dotnet_ws\install\lib\rcldotnet_examples\rcldotnet_talker.bat
- C:\dev\ros2_dotnet_ws\install\lib\rcldotnet_examples\dotnet\rcldotnet_talker.exe
```

So I renamed  the `rcldotnet_talker.bat` and `rcldotnet_listener.bat` to `rcldotnet_talker.bat.bak` and `rcldotnet_listener.bat.bak`.

As a result, everything succeeded.

Thank you!

closed 2020-05-22 by comoc