ros2-dotnet/ros2_dotnet #62: array types

mce16, 2020-05-12:
Hi

I just want to confirm, is there a way to create an array of a primitive type, for instance an array of bytes, or an array of integers etc?

Stepping into the codes for types like std_msgs.msg.UInt16MultiArray or ByteMultiArray seems to point to unsupported types. Is this true?

The usage would be quite limited without the support of those arrays types ..

esteve, 2020-05-12:
@mce16 we are currently working on that. Would you be interested in contributing?

hakuturu583, 2020-05-24:
@esteve Hi, I am integrating ros2-dotnet with Unity and I found same issue.
So, I am very interested in contributing to this issue.

hakuturu583, 2020-05-24:
@esteve how is the current status of the development.

fmrico, 2020-05-24:
Hi @hakuturu583 

I have almost finished this. I can try to PR something soon. Maybe not complete (array of string are a headache)

Best

hakuturu583, 2020-05-25:
@fmrico Great! I am looking forward to see it.
If do you want to some help, feel free to contact me!

hakuturu583, 2020-05-25:
@fmrico 
> array of string are a headache
You mean , your folk of ros2-dotnet cannot generate message like below??

```
string[] array_strings
```

theseankelly, 2020-05-25:
@hakuturu583 that's an overly simplistic view of how language projections go from yaml to C to C#. Let's instead thank @fmrico  for dedicating his free time to enable this feature.

fmrico, 2020-05-29:
Hi @hakuturu583 

I have just PR #68 . It provides support for collections of basic and nested types, except string. Ypu can use 

```
std_msgs/msg/String[] array_strings
```

as workaround. As soon as it will be accepted (I think there will be some iterations due to @esteve and @theseankelly revisions), I will be delighted if you could help me with strings.

I hope it helps!!!

hakuturu583, 2020-06-12:
@fmrico Very Cool, Thank you very much.
I want use it. :+1:

eholmes13, 2020-12-29:
Hello, I've been trying to get the code from PR #68 working on Windows 10 but I'm still unable to get any messages with arrays working. Everything builds just fine, but even after adding the new .dll files to my project in Visual Studio the "data" parameter is still missing for all message types so I am unable to create any messages that contain arrays. Has this happened to anyone else or did I overlook something?

mce16, 2022-04-05:
> Hi @hakuturu583
>
> I have just PR #68 . It provides support for collections of basic and nested types, except string. Ypu can use
>
> ```
> std_msgs/msg/String[] array_strings
> ```
>
> as workaround. As soon [...]

```
int32[] test_int_arr 
std_msgs/String[]  test_string_arr
```
I have the impression above will work, but both doesn't work for me; 
colcon build passed but the types are not generated at all.

hoffmann-stefan, 2022-04-05:
@mce16: which branch where you using?

In the mean time I did an PR improving support for collections (for example collections of strings) that contineus the work from @fmrico, see: https://github.com/ros2-dotnet/ros2_dotnet/pull/89

(you may also be intrested in the follow up PR of mine adding support for services, see #90)

Maybe you should try this branch if you don't have this already.

The packages defining the message types are in the same workspace as described in the README.md? If the packages are in an underlay there is currently no way to generate the C# bindings for that.

mce16, 2022-04-05:
@hoffmann-stefan 
Here is the one,  i will try yours for latest updates, but i am having doubt what have gone wrong since the the branch from @fmrico originally mentioned the support for arrays and the use of std_msgs/String[] as a workaround for string[], but none of the arrays are being generated after build.

https://github.com/fmrico/ros2_dotnet/tree/collections

mce16, 2022-04-06:
updates:

> 
> The packages defining the message types are in the same workspace as described in the README.md? If the packages are in an underlay there is currently no way to generate the C# bindings for that.

My msg definition was an overlay project, and I just tried rebuilding it in the same workspace now as ros2-dotnet and having the same issue again. The only thing different is I am building it against galactic, but I don't think this will pose an issue in this?

hoffmann-stefan, 2022-04-07:
@mce16

The current master branch does only support fixed sized arrays (excuding arrays of strings).
See this example in the code: https://github.com/ros2-dotnet/ros2_dotnet/blob/51dc3803af1b4b5fda0711eed6d58b43013f5aa4/rosidl_generator_dotnet/resource/msg.cs.em#L37-L44
unbounded and bounded arrays are called "sequences" in the idl code.
Didn't think about this initialy as there was some time since I added the support to my PR and since then I only used my feature branch.

So if you update ros2_dotnet to my branch from #89 (or #90) this should work (even without the workaround for string)

As for galactic I did try it on my machine, did work fine (building galactic from souce, creating an ros2_dotnet_ws as in the README.md, creating a new interface package with your msg definition like in the ros2 tutorials). Only used this package in foxy till now.

```
~/ros2_dotnet_galactic_ws
❯ tail build/tutorial_interfaces/rosidl_generator_dotnet/tutorial_interfaces/msg/arrays.cs
        native_destroy_native_message(messageHandle);
    }

    public List<int> Test_int_arr { get; set; }
    public List<std_msgs.msg.String> Test_string_arr { get; set; }
}

}  // namespace msg
}  // namespace tutorial_interfaces

```

mce16, 2022-04-12:
@hoffmann-stefan , perfect, I built it with #90 and finally some progress! The array seems to work now :-)
The only issue i have now is, if I have to build my msg definition modules in the ros2-dotnet workspace instead of overlaying it, it is slow to build the whole ros2-dotnet than just building my msg modules as an overlay whenever I update my msg definitions.

hoffmann-stefan, 2022-04-14:
> @hoffmann-stefan , perfect, I built it with https://github.com/ros2-dotnet/ros2_dotnet/pull/90 and finally some progress! The array seems to work now :-)

Nice to hear 👍

Some thougths about the performance problems:
- Maybe using the colcon package selection arguments helps a little?
  - https://colcon.readthedocs.io/en/released/reference/package-selection-arguments.html
- Building the messages in the same workspace as ros2-dotnet was a step to reduce potential problems while debugging this. Could you try if it now works if the messages are in an higher workspace that soruces the ros2_dotnet one? Like:
  - youre_code_and_msgs_ws
  - ros2_dotnet_ws
  - ros distro

hoffmann-stefan, 2022-11-14:
Closing this as #89 implementing this feature is now merged.