ros2-dotnet/ros2_dotnet #63: Build error on UWP

quattrinili, 2020-05-17:
There is an issue for the UWP build of ros2_dotnet. In particular, the error is on fastrtps and the problem seems to be related to perhaps some differences in libraries and non-compatibility with UWP as it gives some syntax errors:

e.g., error C2332: 'struct': missing tag name (compiling source file ros2-uwp\src\eProsima\Fast-RTPS\src\cpp\transport\TCPAcceptorSecure.cpp)

- this is the command I am running `colcon build --merge-install --packages-ignore rmw_fastrtps_dynamic_cpp rcl_logging_log4cxx rcl_logging_spdlog ros2trace tracetools_launch tracetools_read tracetools_test tracetools_trace --cmake-args -A x64 -DCMAKE_SYSTEM_NAME=WindowsStore -DCMAKE_SYSTEM_VERSION=10.0.17134 -DTHIRDPARTY=ON -DINSTALL_EXAMPLES=OFF -DBUILD_TESTING=OFF -DRCL_LOGGING_IMPLEMENTATION=rcl_logging_noop  --event-handlers console_cohesion+`
- I am running this with the repos file for UWP? https://github.com/ros2-dotnet/ros2_dotnet/blob/master/ros2_dotnet_uwp.repos
- I am compiling on Windows 10 with Visual Studio 2017.

Note that the Desktop ros2_dotnet compiles fine. My sense is that the problem might be on the ASIO library, and I tried to get the uwp version with vcpkg .12.2-2, but still without success.

(Split from comments on issue #55)

mcevoyandy, 2020-05-19:
@quattrinili were you able to resolve this issue? I've got the same problem.

theseankelly, 2020-05-20:
Hey @quattrinili and @mcevoyandy 

Quick thought from my side -- have you used ROS2 on Windows before (and thus, have you installed all the nuget package dependencies necessary for the Desktop install?). I've noticed that if cmake can find them, then ASIO will not source its own copies, and you'll end up pulling in the non-UWP libraries to the build which can cause build and/or runtime errors.

Uninstall all ROS2 third party dependencies that the ROS2 installation instructions had to install (libasio, tinyxml, log4cxx, etc). The repos files (for UWP) are completely standalone, and the `-DTHIRDPARTY=ON` flag is specifically for FastRTPS - it checks out and builds its own copy of all dependent libraries. Doing it this way ensures the dependencies are also UWP ready.

theseankelly, 2020-05-20:
Oh and uninstall OpenSSL as well -- my memory is foggy but I believe if OpenSSL is not present, FastRTPS will not build with security enabled (and your build error is in something called "TCPAcceptorSecure")

esteve, 2020-05-20:
@quattrinili @mcevoyandy yes, @theseankelly is right, uninstall OpenSSL for now. The reason for this failing is that `TCPAcceptorSecure` uses `interface` as a variable name, but that is defined in `windows.h` as a `typedef` for something else. However, because we don't install OpenSSL in CI, we didn't see this issue.

I'm updating our fork of Fast-RTPS to not build the security stuff if targeting UWP.

esteve, 2020-05-20:
I've disabled support for the security extensions in FastRTPS if building for UWP in https://github.com/ros2-dotnet/Fast-RTPS/commit/40072bb821e567d96e389d6180960416257a7c0e

@quattrinili @mcevoyandy could you try again with an updated copy of our fork of FastRTPS? 

Thanks!

mcevoyandy, 2020-05-20:
Thanks for the fast response! This enabled fastrtps to build properly :)

Here's the steps I followed:
1. Windows > Add or Remove Programs > OpenSSL > uninstall
2. in cmd: > choco uninstall asio
3. Remove build, install, log folders from previous build attempts
4. Repeated install instructions

quattrinili, 2020-05-23:
I forgot that I had OpenSSL and I uninstalled it. The error related to it disappeared. However, unfortunately, I am still having problems, this time with linking with a link with foonathan_memory. I am digging into it.

`foonathan_memory-0.6.2.lib(virtual_memory.obj) : error LNK2019: unresolved external symbol __imp_VirtualAlloc referenced in function "void * __cdecl foonathan::memory::traits_detail::allocate_node<class foonathan::memory::virtual_memory_allocator>(struct foonathan::memory::traits_detail::full_concept,class foonathan::memory::virtual_memory_allocator &,unsigned __int64,unsigned __int64)" (??$allocate_node@Vvirtual_memory_allocator@memory@foonathan@@@traits_detail@memory@foonathan@@YAPEAXUfull_concept@012@AEAVvirtual_memory_allocator@12@_K2@Z) `

quattrinili, 2020-05-25:
Ok, found the problem: it is in the foonathan/memory, which uses `VirtualAlloc`, however UWP does not support it. Instead, `VirtualAllocFromApp` should be used, and that works also for Windows Desktop app (see https://docs.microsoft.com/en-us/windows/win32/api/memoryapi/nf-memoryapi-virtualallocfromapp). I modified it in my fork https://github.com/quattrinili/memory -- I plan to submit a pull request for foonathan/memory, but you are welcome to use it if you would like in the meanwhile.

This allowed me to fully compile ros2_dotnet for UWP. I tried with a talker and a listener in a simple UWP app, and at least right now it is not crashing; though for some reasons, they are not able to communicate. I still get some DLLNotFoundException on rcl_dotnet_common. I will check a couple of other things.

esteve, 2020-05-25:
@quattrinili 

> This allowed me to fully compile ros2_dotnet for UWP

That's great! Fast-RTPS introduced `foonathan_memory` as a dependency, but it's weird that our CI hasn't caught that. In any case. thanks! :-)

quattrinili, 2020-05-28:
Maybe it was working with a previous version of MSVC? I submitted the pull request upstream for foonathan memory, but I see that it fails with MSVC 14 (should be Visual Studio 2015). On my system, both VS 2017 and 2019 work. All these versions make it hard to ensure things work properly... :)