ros2-dotnet/ros2_dotnet #64: How can use it in the unity under Linux?

Andor233, 2020-05-19:
I'm a developer who is creating a simulation environment for Robot. Our robot is running under Linux, So How can use it in the unity under Linux?

adamdbrw, 2020-05-19:
You would need some additional effort for that. Assemblies and native libraries (dll and so files) built by this project need to be in proper Plugin directories of Unity. Additionally, FastRTPS and UnityPlayer libraries conflict for current version, but you can make it work with CycloneDDS.
There is a couple of important things missing still, though the project is on a good way to fill the gaps (I am taking about array types in message generation such as you would want for PointCloud2 ros2 messages coming from your lidar).

Since we are using such Unity plugin with full message support for quite a while already in our company (Robotec.AI), I could help you go through the process. Also, if you need a demo, just let me know. But also you can wait for us to release it open source along with our ros2 C# project that was built based on a fork of ros2_dotnet. I hope to contribute back to this project once we get a release, especially because it picked the pace and has a good leadership.

Andor233, 2020-05-22:
My unity version is 2020, and it doesn't conflict with FastRTPS. It seems like works fine.
By the way, how can I use custom message?

adamdbrw, 2020-05-29:
You need to run cs generator with these messages - easiest way is to drop the message package into build tree and add it to dependencies. The register hooks for the generator will take care of the rest

PaoloTo, 2020-09-08:
Dear @adamdbrw, there are updates on unity implementation of ros2_dotnet?