ros2-dotnet/ros2_dotnet #67: DLL not found in UWP blank app

quattrinili, 2020-05-28:
After being able to compile for UWP (see #63), now I am trying to have two simple UWP blank programs, one listener, one talker. Following the instructions to add the dlls and the references, the run of such simple programs report error in loading DLLs. (Note that the desktop version and the corresponding examples work on my system.)

I looked at a previous issue that seemed to be related to issue #2 (and specifically a comment on how to debug it https://github.com/ros2-dotnet/ros2_dotnet/issues/2#issuecomment-435695184), but the suggestions provided there didn't work for me.

I tried all combinations of the following: 
  - Win32 and x64 compilation for ros2dotnet_uwp; 
  - Visual Studio 2017 and 2019 (and accordingly, .NET CORE 2.1 vs 3)
  - "Compile with .NET Native tool chain" enabled or not.
After using gflags.exe, when .NET Native tool chain is enabled, I can see directly in Visual Studio output that some DLLs are unable to load; in particular SharedLibrary.dll, and another time wintypes.dll. I ensured that WindowsApps folder is accessible changing its properties and I tried to add explicitly them in the project, and changing the properties similarly as for other DLLs without success.

Without that flag, I cannot directly see it from Visual Studio (and running an app via commandline requires some extra steps that I didn't partake yet), and the error is a generic DLL not found for rcldotnet_common.dll. Looking at the dependency graph of lib\rcldotnet_common.dll this is the high level view of the modules:

[image]

Any suggestion is welcome: it would be beneficial to have a working UWP blank app example similarly to the desktop version. If I succeed, happy to provide one.

mcevoyandy, 2020-05-28:
I'm having similar issues but also find difficulty in applying the suggestions in #2.

quattrinili, 2020-05-28:
Indeed it looks similar to #65, but I don't have Unity in my case (that's the reason why I opened a new issue. The configuration is slightly different, but it shouldn't change much -- hopefully the solution if we find it will be applicable to both cases!

mcevoyandy, 2020-05-28:
Agreed there are slight differences, but I think the underlying issue is the same.

I also have desktop examples working, but get missing references to MSVC140 in Dependency Walker on some dlls...  so now have VS 2019, 2017, and 2015 installed. You're right in the other thread. This is confusing to get working :)

mcevoyandy, 2020-05-28:
@quattrinili I don't get any dependency errors on rcldotnet_common, but do get these missing dependencies on `rcldotnet_native.dll`

[image]

quattrinili, 2020-06-09:
I finally had a little bit of time to take a stab again at this and have some updates. I think that  .NET Native tool chain should be disabled -- I suspect that there are parts of the code of ros2_dotnet that are not compatible with that.

With that option disabled, the error that appears in the console "Exception thrown: 'System.DllNotFoundException' in rcldotnet_common.dll" is coming from the CheckPlatform(), i.e., https://github.com/ros2-dotnet/ros2_dotnet/blob/master/rcldotnet_common/DllLoadUtils.cs#L72. In particular, the attempts in from IsUnix() and IsMacOSX(). Afterwards, it is able to load rcldotnet_native.dll -- despite the missing DLLs as shown by the DependencyGraph. Note that even if UWP app, the IsWindowsDesktop() will be executed when tested on your laptop. I tried also to swap UWP with the WindowsDesktop(), and the UWP loading is called and seems to work.

The problem I have now is that, assuming that the DLLs are loaded properly, while in the same app, publisher and subscriber are exchanging messages, a different app with the publisher code, and another UWP app with the listener code don't seem to talk to each other. I guess it is something related to the way ROS2 environment is created. 

I think we can close at least the issue about DLL not found, unless @mcevoyandy has still some other comments related to this, but any suggestions on how to set the ROS2 environment of the UWP app so that it is visible outside of the app itself are welcome.

theseankelly, 2020-06-09:
The .NET Native issue is known (see issue #39) -- sorry it did not come back to my memory when reading this thread sooner.