ros2-dotnet/ros2_dotnet #75: Build error on win 10

michaelchi08, 2020-10-19:
Win10 and .NET Core 3.1 is installed

Starting >>> rcldotnet_common
--- stderr: rcldotnet_common
CMake Warning (dev) at C:/opt/ros/foxy/x64/Lib/site-packages/cmake/data/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args`
  (DOTNET_CORE) does not match the name of the calling package (DotNetCore).
  This can lead to problems in calling code that expects `find_package`
  result variables (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/dev/ros2_dotnet_ws/install/share/dotnet_cmake_module/cmake/Modules/dotnet/FindDotNetCore.cmake:37 (find_package_handle_standard_args)
  C:/dev/ros2_dotnet_ws/install/share/dotnet_cmake_module/cmake/Modules/FindCSBuild.cmake:20 (find_package)
  C:/dev/ros2_dotnet_ws/install/share/dotnet_cmake_module/cmake/Modules/FindDotNETExtra.cmake:15 (find_package)
  CMakeLists.txt:9 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Failed   <<< rcldotnet_common [8.31s, exited with code 1]
Aborted  <<< rosidl_generator_c [10.1s]

Summary: 7 packages finished [28.3s]
  1 package failed: rcldotnet_common
  1 package aborted: rosidl_generator_c
  1 package had stderr output: rcldotnet_common
  28 packages not processed

Anybody knows what's going on and how to fix this?
Thanks

berkayalpcakal, 2020-11-22:
Hi @michaelchi08,

I am also facing the same error. rcldotnet_common fails to be built without any explicit error log.
Have you resolved the issue?

michaelchi08, 2020-11-22:
Hi @berkayalpcakal , no unfortunately not yet 😞

berkayalpcakal, 2020-11-24:
Hi again, 

I was able to build this repository in a ros2 workspace on my Ubuntu 18.04 machine. Then I wanted to build these packages in a ros2 workspace on my Win 10 machine. But 'rcldotnet_common' does not build for some reasons, and gives the aforementioned build output.

Then I decided to follow the way as described in [here](https://github.com/DynoRobotics/UnityRos2#building-custom-interfacesmessages) (using Linux kernel based docker on my Win 10 machine). Finally I successfully built the ros2dotnet on Win 10.

michaelchi08, 2020-11-27:
Hi, thanks for letting me know. I'll give it a try. it'll be nice if we could run this natively though. Do yo know if it's possible to run just this in docker and the all other ros2 nodes natively on windows?

JackHaleGreenfield, 2023-08-08:
Hello. I tried building ros2_dotnet on my windows 10 system today (with Foxy release of ROS2 installed), and I have been seeing the same errors previously mentioned here. 

I decided to run the colcon build command with an extra flag (`colcon build --merge-install --event-handlers console_cohesion+`) and I started seeing some new output in error messaging:

`C:\ros2-windows\include\rcl\rcl/event_callback.h(18,10): fatal error C1083: Cannot open include file: 'rmw/event_callback_type.h': No such file or directory (compiling source file C:\Users\Jack Hate\Desktop\GetStartedApp\ros2_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet\rcldotnet_node.c) [C:\Users\Jack Hate\Desktop\GetStartedApp\ros2_dotnet_ws\build\rcldotnet\rcldotnet_native.vcxproj]`

So then I poked around the install location for ROS2 and I noticed that `event_callback_type.h` does in fact exist but it's path is `rmw\rmw\event_callback_type.h` when starting from the `include` folder.

To verify this was the issue, I tried moving the event_callback_type.h file to the parent directory. Doing this still resulted in a build failure but with new error messages that appear to be syntax issues: `C:\ros2-windows\include\rcl\rcl/subscription.h(59,1): error C2059: syntax error: '}'`

See attached file for a larger snapshot of the build output.
[ros2_dotnet_build_output.txt](https://github.com/ros2-dotnet/ros2_dotnet/files/12296021/ros2_dotnet_build_output.txt)

I am out of my depth in debugging this further, but maybe someone else can figure out what's going wrong from my findings.