ros2-dotnet/ros2_dotnet #77: Build error on ros2 foxy untuntu 20.04 with .netcore

michaelchi08, 2020-11-30:
Finished <<< std_srvs [1.12s]                                               
--- stderr: unique_identifier_msgs                                          
/home/ros/ros2_dotnet_ws/build/unique_identifier_msgs/rosidl_generator_dotnet/unique_identifier_msgs/msg/uuid.c:11:10: fatal error: rosidl_generator_c/message_type_support_struct.h: No such file or directory
   11 | #include "rosidl_generator_c/message_type_support_struct.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/unique_identifier_msgs__dotnetext.dir/build.make:82: CMakeFiles/unique_identifier_msgs__dotnetext.dir/rosidl_generator_dotnet/unique_identifier_msgs/msg/uuid.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:191: CMakeFiles/unique_identifier_msgs__dotnetext.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
---
Failed   <<< unique_identifier_msgs [2.56s, exited with code 2]
Aborted  <<< builtin_interfaces [2.65s]          
Aborted  <<< lifecycle_msgs [2.87s]                                 

Summary: 8 packages finished [5.83s]
  1 package failed: unique_identifier_msgs
  2 packages aborted: builtin_interfaces lifecycle_msgs
  3 packages had stderr output: builtin_interfaces lifecycle_msgs unique_identifier_msgs
  18 packages not processed

clightcap-kcl, 2020-11-30:
@michaelchi08 I just ran into the same issue a few minutes before your post. I believe it may be related to ROS2 Foxy no longer using the rosidl_generator_c namespace:
https://index.ros.org/doc/ros2/Releases/Release-Foxy-Fitzroy/#rosidl-generator-c-cpp-namespace-api-changes

Has anyone successfully built this repo with ROS2 Foxy?

fmrico, 2020-12-02:
Hi

Have a look to #68 . In this PR I have fixed this error.

Best

hoffmann-stefan, 2023-04-15:
Close as fixed as branch from fmrico is merged.

closed 2023-04-15 by hoffmann-stefan