ros2-dotnet/ros2_dotnet #78: Build overlaying Galactic

mce16, 2021-07-11:
I an trying my luck with building it overlaying galactic and i encounter an error in file path like below
```

Failed   <<< rosidl_typesupport_introspection_c [7.89s, exited with code 1]

C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.CppBuild.targets(321,5): error MSB3491: Could not write lines to file "x64\Release\ament_cmake_python_symlink_rosidl_typesupport_introspection_c_setup\ament_cm.81E1E29E.tlog\ament_cmake_python_symlink_rosidl_typesupport_introspection_c_setup.lastbuildstate". The specified path, file name, or both are too long. The fully qualified file name must be less than 260 characters, and the directory name must be less than 248 characters. [D:\MyGit\ros2-dotnet\ros2_dotnet_ws\build\rosidl_typesupport_introspection_c\ament_cmake_python_symlink_rosidl_typesupport_introspection_c_setup.vcxproj]

```

mce16, 2021-07-11:
Latest update: Successfully built as overlay on galactic release with branch from @fmrico , with vs2019 x64

ooeygui, 2021-10-17:
https://github.com/ros2-dotnet/ros2_dotnet/pull/84 should enable this; but it hasn't been tested.

hoffmann-stefan, 2023-04-15:
Closing this issue as the galactic isn't oficially supported any longer, the branch from fmrico is already merged and the author was successfull in getting it to work.

closed 2023-04-15 by hoffmann-stefan