ros2-dotnet/ros2_dotnet #80: [Feature] Support Humble

ooeygui, 2021-10-11:
This work item is to support the Humble release of ROS2.

hoffmann-stefan, 2023-04-22:
As of https://github.com/ros2-dotnet/ros2_dotnet/pull/98 support for Humble works on Linux, Windows Desktop passes CI, but haven't tested it yet on my Laptop as I don't (yet) have a ROS setup on Windows itself (Using WSL).

If someone could report that this works we can close this issue.

timonmentink, 2024-07-22:
After installing ROS2 humble on windows, ros2-dotnet can be build for humble with the following commands
```
call \dev\ros2_foxy\local_setup.bat
md \dev\ros2_dotnet_ws\src
cd \dev\ros2_dotnet_ws
curl -sk https://raw.githubusercontent.com/ros2-dotnet/ros2_dotnet/main/ros2_dotnet_humble.repos -o ros2_dotnet_humble.repos
vcs import \dev\ros2_dotnet_ws\src < ros2_dotnet_humble.repos
colcon build --merge-install
```

hoffmann-stefan, 2024-07-23:
@timonmentink: Thank you for testing this and getting it to work with #126 :)