ros2-dotnet/ros2_dotnet #91: Reconsider property naming convention in generated messages

hoffmann-stefan, 2022-01-04:
While programming using this library I regulary stumble upon the generated names of properties in the message types. For example in the [header message](https://docs.ros2.org/foxy/api/std_msgs/msg/Header.html) the field `frame_id` gets converted to `Header.Frame_id`.

`Frame_id` is neither left nor right:
- It isn't the name defined in the `*.msg` file and what c++ and python uses.
- It isn't the C#/.NET naming convention for public properties (PascalCase).

Posible alternatives to choose:

- Don't modify at all and use the casing as in the `.msg` files: `frame_id`
  - If the other options are to much hassle.
  - Makes it easier to search the field name across packages written in different languages.
- Fully convert to PascalCase to feel good for the .NET developer: `FrameId`
  - Does make C# code look nice, as petty much every library uses this convention. Otherwise there would be mixing of naming conventions in the C# code base.
- At least upercase each word, but keep the underscors: `Frame_Id`
  - Would still allow search across diffenrent languages when configured to ignore case.

My favorite would be CamelCase, the next would be don't change the name at all and use what is defined in the `*.msg` file.

Are there other alternatives or things to consider?

Related to this would be the names of the request/response classes of services (`SetBool_Request`/`SetBool_Response`) and the general naming convention of the public namespaces/types/members in this library (`RCLDotnet` vs. `RclDotnet`), but i think this should be seperate issues.

esteve, 2022-01-11:
@hoffmann-stefan that's a great idea! I prefer 2 as well, PascalCase seems the most idiomatic for C# Would you be interested in submitting a PR for this? Thanks.

hoffmann-stefan, 2022-01-11:
@esteve Yes, I would like to submit a PR for this. Altough this should be done after the other larger open PRs are merged, as this would result in somewhat huge merge conflicts if done before those are merged.

shschaefer, 2022-01-12:
I would suggest option #1 as it is consistent with message identifiers across all platforms.  It is important that the C# language interfaces are idiomatic, but will cause debugging inconsistencies at the messaging layer when this convention is changed.

hoffmann-stefan, 2022-06-15:
Seems like `ros2_java` is using CamelCase as well: https://github.com/ros2-java/ros2_java/blob/434e6f55253bfe2cb9ce34799fe548bbf4998d0e/rosidl_generator_java/rosidl_generator_java/__init__.py#L28-L30
Opend a PR for this: #93

closed 2023-04-22 by hoffmann-stefan