ros2-dotnet/ros2_dotnet #92: Reenable UWP Builds for Foxy

ooeygui, 2022-04-21:
The ROS2.net UWP builds are not functioning due to ROS2 itself failing to build on the Universal Windows Platform. An effort is under way to upstream ROS2 for UWP, after which time the ROS2.net UWP builds can be reenabled. 

Here are the tracking repositories for the various ROS2 on UWP changes:
https://github.com/ms-iot/Fast-DDS.git
https://github.com/ms-iot/rcpputils.git
https://github.com/ms-iot/rcutils.git
https://github.com/ms-iot/libyaml_vendor.git
https://github.com/ms-iot/poco_vendor.git

ooeygui, 2022-04-21:
During the process of upstreaming these changes, it was discovered that poco_vendor has been archived, instead relying on the OS libraries. Poco itself has not been [UWP enabled](https://github.com/pocoproject/poco/issues/2544). UWP has been grafted on - including with poco_vendor and vcpkg.

@esteve I'd like to propose creating a fork of  https://github.com/ros2/poco_vendor in the ros2_dotnet organization, which can host poco for UWP changes which are currently in ms-iot until official support is available. I'd also like input on alternative approaches.

ooeygui, 2022-05-09:
First PRs are pulled into humble. Working on next batch.

esteve, 2022-05-10:
@ooeygui no problem about the poco_vendor fork, but it seems that https://github.com/ros2/poco_vendor has been archived. Is it still being used by ROS2? I think the only part of POCO in use was for dynamically loading a library, which was replaced by calling `dlopen` / `LoadLibrary` directly.

ooeygui, 2022-05-10:
@esteve Let me check on that.

ronwalf, 2023-11-06:
@ooeygui Are there more details about what's blocking UWP builds, and what would be necessary for builds on Humble and later?

ooeygui, 2023-11-06:
Hi @ronwalf, 
I wish I could continue to work on this.

There are core changes in ROS and the Middleware that make them compatible with the Hololens 2 app container model (which is different than the Hololens 1 app container). I've made those changes in the ms-iot forks, but I have not propagated them upstream into rolling. The code in the forks is "Proof of concept" quality, so need to have effort put into unit tests, and multiplaform testing before they can be PR'd into their respective repos. I don't know what the delta is to migrate the changes to current releases.