ros2-dotnet/ros2_dotnet #117: warning MSB8065: from Empty arrays.hpp.rule

InfoShellTex, 2023-07-20:
Building on windows 10, using visual studio 2019 developer terminal with admin privileges', i built/installed ros2 foxy via the [aka.ms way here](https://ms-iot.github.io/ROSOnWindows/GettingStarted/SetupRos2.html)
the package "rosidl_generator_cpp" fails and gives this output:
`-- Selecting Windows SDK version 10.0.22000.0 to target Windows 10.0.19045.
-- Found ament_cmake: 0.9.9 (C:/opt/ros/foxy/x64/share/ament_cmake/cmake)
-- Found PythonInterp: C:/opt/ros/foxy/x64/python.exe (found suitable version "3.8.3", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: C:/opt/ros/foxy/x64/python.exe
-- Override CMake install command with custom implementation using symlinks instead of copying resources
-- Found ament_lint_auto: 0.9.6 (C:/opt/ros/foxy/x64/share/ament_lint_auto/cmake)
-- Found rosidl_runtime_c: 1.3.1 (C:/ProgramFilesAlt/Ros2Workspaces/ros2_dotnet/install/share/rosidl_runtime_c/cmake)
-- Found rosidl_runtime_cpp: 1.3.1 (C:/ProgramFilesAlt/Ros2Workspaces/ros2_dotnet/install/share/rosidl_runtime_cpp/cmake)
-- Found test_interface_files: 0.8.1 (C:/ProgramFilesAlt/Ros2Workspaces/ros2_dotnet/install/share/test_interface_files/cmake)
-- Found rosidl_adapter: 1.3.1 (C:/ProgramFilesAlt/Ros2Workspaces/ros2_dotnet/install/share/rosidl_adapter/cmake)
-- Found PythonInterp: C:/opt/ros/foxy/x64/python.exe (found version "3.8.3") 
-- Found gtest sources under 'C:/opt/ros/foxy/x64/src/gtest_vendor': C++ tests using 'Google Test' will be built
-- Added test 'copyright' to check source files copyright and LICENSE
-- Added test 'cppcheck' to perform static code analysis on C / C++ code
-- Configured cppcheck include dirs: 
-- Configured cppcheck exclude dirs and/or files: 
-- Added test 'cpplint' to check C / C++ code against the Google style
-- Configured cpplint exclude dirs and/or files: 
-- Added test 'flake8' to check Python code syntax and style conventions
-- Added test 'lint_cmake' to check CMake code style
-- Added test 'pep257' to check Python code against some of the style conventions in PEP 257
-- Added test 'uncrustify' to check C / C++ code style
-- Configured uncrustify additional arguments: 
-- Added test 'xmllint' to check XML markup files
-- Configuring done
-- Generating done
-- Build files have been written to: C:/ProgramFilesAlt/Ros2Workspaces/ros2_dotnet/build/rosidl_generator_cpp
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

  Generating C++ code for ROS interfaces
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\arrays__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\basic_types__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\bounded_sequences__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\constants__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\defaults__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\empty__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\multi_nested__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\nested__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp__cpp.vcxproj]
C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(241,5): warning MSB8065: Custom build for item "C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\CMakeFiles\d2d3d8c6d6b556e59d81f60ee3332b51\arrays.hpp.rule" succeeded, but specified output "c:\programfilesalt\ros2workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp\msg\detail\strings__type_support.hpp" has not been created. This may cause incremental build to work incorrectly. [C:\ProgramFilesAlt\Ros2Workspaces\r [... cut ...]
C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\rosidl_generator_cpp\rosidl_generator_cpp/msg/empty.hpp(10,10): fatal error C1083: Cannot open include file: 'rosidl_generator_cpp/msg/detail/empty__type_support.hpp': No such file or directory [C:\ProgramFilesAlt\Ros2Workspaces\ros2_dotnet\build\rosidl_generator_cpp\test_traits.vcxproj]
`

I check the arrays.hpp.rule file and the contents is just this line: "# generated from CMake"
thats it, what generates that file and how would i go about troubleshotting it, im not sure what if thats expected or normal.

hoffmann-stefan, 2023-08-15:
Hi @InfoShellTex,

sorry for the late reply, but was busy at work and spare time recently.

I don't have a working windows ROS2 dotnet setup myself yet. I started setting it up, but all those manual install steps did take longer than I thought last time I started this, since then I haven't continued.
So I'm sorry I can't really dig into this right now, but I will look at this once I have that setup again.

btw. Wat was the reason to close your other Issue (https://github.com/ros2-dotnet/ros2_dotnet/issues/115)? Did you resolve it? Maybe you could write a short note what to did to resolve it for others that stumble upon this later on.

InfoShellTex, 2023-08-17:
I did not resolve it. I closed it because I am now building off of the foxy version, since i can successfully install it, where the #115 was using humble (from binaries), to me, didnt feel like a complete install compared to foxy. 
And i didn't want to have many things going on at once. But Im fine with it being reopened as that issue still does exist.

And thanks, no worries! any help is better than none at this point.