ros2-dotnet/ros2_dotnet #119: fail to run rcldotnet_examples on win10 OS，throw System.ArgumentNullException

cqulpj, 2023-09-17:
my PC system is win10, the ros2 version is foxy, visual studio 2019, .net 3.1/6/7, first, every steps of build was ok, but when I try run the examples, it throw Exceptions, the error messages as follow:

PS C:\opt\ros_dotnet_ws> .\install\local_setup.ps1
PS C:\opt\ros_dotnet_ws> ros2 run rcldotnet_examples rcldotnet_talker

Unhandled Exception: System.TypeInitializationException: The type initializer for 'ROS2.RCLdotnetDelegates' threw an exception. ---> System.ArgumentNullException: Value cannot be null.
Parameter name: ptr
   at System.Runtime.InteropServices.Marshal.GetDelegateForFunctionPointer(IntPtr ptr, Type t)
   at ROS2.RCLdotnetDelegates..cctor() in C:\opt\ros_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet\RCLdotnet.cs:line 451
   --- End of inner exception stack trace ---
   at ROS2.RCLdotnet.Init() in C:\opt\ros_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet\RCLdotnet.cs:line 1399
   at ConsoleApplication.RCLDotnetTalker.Main(String[] args) in C:\opt\ros_dotnet_ws\src\ros2_dotnet\ros2_dotnet\rcldotnet_examples\RCLDotnetTalker.cs:line 12

who can give me some suggestions, well regards.

SiebeVerhoeven, 2023-11-13:
I'm running against the same problem, is there anyone that is able to help?

hoffmann-stefan, 2023-12-05:
@cqulpj, @SiebeVerhoeven: Sorry for the late reply, currently my "spare" time I can allocate for this project is used to get the open PRs done. I tried to setup ros2_dotnet on windows once, but didn't have time to continue yet. 

@SiebeVerhoeven did you try a cleen rebuild? Those functions should be defined in the c part of the library, so not sure how they would be missing for you.