ros2-dotnet/ros2_dotnet #137: About additional message packages

WYihei, 2024-12-18:
Is it possible to add additional message packages to ros2_dotnet to generate the messages I need, such as moveit_msgs?

hoffmann-stefan, 2024-12-20:
Hi @WYihei,

yes this is possible. Like in the Readme check out all repos needed to build ros2_dotnet and the default message packages. Afterwards you can add additional massage repositories/packages to the workspace.

You can't use the message packages installed via system packages, but you can clone the repositories with the right branch for your ROS distro into the workspace.

Please note that you need to add all dependencies of the message packages as well.

For MoveIt messages this should be something like:
```
cd $ROS2_DOTNET_WS/src/
git clone -b <branch> https://github.com/moveit/moveit_msgs.git
git clone -b <branch> https://github.com/OctoMap/octomap_msgs.git
git clone -b <branch> https://github.com/wg-perception/object_recognition_msgs.git
# hope I got all dependencies
```

WYihei, 2024-12-22:
> Hi @WYihei,
>
> yes this is possible. Like in the Readme check out all repos needed to build ros2_dotnet and the default message packages. Afterwards you can add additional massage [...]

Thank you so much! I messed up earlier because I didn’t realize I needed to clone these two repositories.

closed 2024-12-22 by WYihei